Ultimaker/Cura #11284: Cura Slicer dropping out, and menu items not responding

Jax3DTrains, 2022-01-17:
### Application Version

4.1.3

### Platform

Ubuntu

### Printer

Snapmaker A350T

### Reproduction steps

1. clicked on menu choice, and the window does not open.
2. Rebooted with no success, removed program and redownloaded with no success.
3. Program also closes in the middle of connfiguring a file.

### Actual results

Problem still happens.

### Expected results

I would have hoped it would fix the issue by rebooting and also reloading the program, but here I am reporting to you.
Can't show you a project file that shows the bug, because it happens when I try to use the program. Kind of hard to show a menu item not opening.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I can't show in a file the program shutting down all of a suden. I also can't show you in a project file how the menu choices aren't responding to being clicked. All I can do is report it, and hope there is a fix.

Thanks

Jax3DTrains, 2022-01-18:
Here is a screenshot of the issue that just happened with the menu choices. Some choices are active, but others are not. With this issue, two menu choices came up.
[image]

fvrmr, 2022-01-18:
Hi @Jax3DTrains thank you for your report.
Are you using the our AppImage?

Jax3DTrains, 2022-01-18:
Thanks Fenne,

Yes I am. Here is the name on the file I downloaded and installed on Ubuntu.
Ultimaker_Cura-4.13.0.AppImage

Ghostkeeper, 2022-01-25:
The choices that are active are purposeful, indeed. You can't "Multiply Selected" since there is nothing selected.

I think we'll need a log file to know more. It is in the bug reporting template to provide a log file. Perhaps you could add it there?

I haven't been able to reproduce this issue so far. I can right click anywhere to open a menu, and the application menu at the top functions fine for me, both on Ubuntu 20.04 and on 21.10.

closed 2022-02-08 by no-response[bot]