Ultimaker/Cura #11289: Temperature settings not set properly when G29 is in printer start code.

rongoodman303, 2022-01-18:
### Application Version

4.13

### Platform

Windows 10

### Printer

Wanhao I3 Plus

### Reproduction steps

Printer start code:

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
M204 P500 ; PETG Acceleration
M205 J0.30 ; PETG Junction Deviation
;M851 Z-3.00 ; Force z offset if needed.
G28 ;Home all
G29; ABL Start. Mesh will be auto saved by TFT.
G92 E0 ;zero the extruded length
G1 Z3.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.3 Y20 Z0.24 F5000.0 ; Move to start position
G1 X0.3 Y140.0 Z0.24 F1500.0 E15 ; Draw the first line
G1 X0.8 Y140.0 Z0.24 F5000.0 ; Move to side a little
G1 X0.8 Y20 Z0.24 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
;Put printing message on LCD screen
M117 Printing...

=======================================

Generated Gcode:

;FLAVOR:Marlin
;TIME:20568
;Filament used: 14.1926m
;Layer height: 0.2
;MINX:58.8
;MINY:56.008
;MINZ:0.24
;MAXX:141.2
;MAXY:143.992
;MAXZ:32.04
;Generated with Cura_SteamEngine 4.13.0
M140 S65
M105
M190 S65
M104 S230
M105
M109 S230
M82 ;absolute extrusion mode
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
M204 P500 ; PETG Acceleration
M205 J0.30 ; PETG Junction Deviation
;M851 Z-3.00 ; Force z offset if needed.
G28 ;Home all
G29; ABL Start. Mesh will be auto saved by TFT.
; Move part temp sets to here
G1 Z15.0 F200.0 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 Z3.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.3 Y20 Z0.24 F5000.0 ; Move to start position
G1 X0.3 Y140.0 Z0.24 F1500.0 E15 ; Draw the first line
G1 X0.8 Y140.0 Z0.24 F5000.0 ; Move to side a little
G1 X0.8 Y20 Z0.24 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
;Put printing message on LCD screen
M117 Printing...
... etc

Note that the temperature settings are placed at the top of the gcode file, before the printer start commands.

In Marlin: 
#define PREHEAT_BEFORE_LEVELING 
#if ENABLED(PREHEAT_BEFORE_LEVELING) // Adjust if your normal material needs something else.
  #define LEVELING_NOZZLE_TEMP 160   // (°C) Only applies to E0 at this time
  #define LEVELING_BED_TEMP     65
#endif

Issue can be replicated with the gcode fragment included here.

### Actual results

Setting temperatures before inserting printer start commands may work if there is no BLTouch or other ABL in place where Marlin is configured to preheat the nozzle and bed to certain temps before leveling, as part of the G29 process.  In that case, the temperatures previously set are voided and replaced by the new temperatures that are part of the G29 process, and they are never restored after the G29 ABL process finishes. The print, including prin commands in the printer startup settings, tries to start at the G29 set temperatures, which are not correct per the Cura profile settings for that job. The print fails, or may run but at the wrong temperatures.

### Expected results

The printing of the nozzle purge/test lines that follow the G29 should start with the correct temperatures previously set at the top of the gcode file. 
Temperature settings need to be restored by insertion of the specified project/part temperatures after any G29 in the printer start commands, if there is a G29 within the printer start commands.  

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

This behavior can be replicated with just the gcode fragment provided above, as long as it is tested on a printer with BLTouch or other ABL device installed, and with Marlin settings similar to the above, with a low nozzle temp for bed leveling
[WDI3P_Mini_Monster_Willy_Frame_LG.3mf.txt](https://github.com/Ultimaker/Cura/files/7892732/WDI3P_Mini_Monster_Willy_Frame_LG.3mf.txt)
.

GregValiant, 2022-01-19:
The StartUp G-Code is from your printer definition file that was put together by a contributor to Cura or by Wanhao.

You need to add a couple of lines.  The code below will start to heat the bed and then heat the hot end to 150 and wait for it and then wait for the bed temperature.  Then the printer will auto-home and the ABL will run and hopefully not ooze.  The last added line will heat the hot end to the Initial Layer Print temperature and wait for it to get there.
M205 J0.30 ; PETG Junction Deviation
;M851 Z-3.00 ; Force z offset if needed.
**M140 S{material_bed_temperature_layer_0} ;start to heat the bed
M109 S150 ;warm up the hot end
M190 S{material_bed_temperature_layer_0} ;wait for the bed to finish heating to initial layer bed temperature** 
G28 ;Home all
G29; ABL Start. Mesh will be auto saved by TFT.
**M109 S{material_print_temperature_layer_0} ;heat the hot end to the initial layer print temperature and wait for it**
G92 E0 ;zero the extruded length
G1 Z3.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

When Cura sees the keywords in the Startup Gcode it will not add the temperatures to the beginning and so that code should work.  You may want to put a semi-colon in front of the "move over, draw the line, etc." lines.  They are the purge lines and if you are using a large skirt or brim you may not need them.  I leave them in unless I am printing something wide that is going to run over them and then I just open the gcode and comment them out with the semi-colon.

rongoodman303, 2022-01-19:
Greg, Thanks. That helped and fixed the issue for now.

However, I would still like to see Cura handle this more automatically, perhaps by adding the project specified layer 0 bed and hotend temps before and after a G29, if there is a G29 and there are not already user defined temp settings in the printer start code.  

I changed the title, but can't seem to change the tags on this from Bug to Feature Request. Can a moderator or admin do that please?

GregValiant, 2022-01-20:
Maybe there is something they will do but personally I don't see it as a Cura thing.  I see it as a Manufacturer/Contributor thing.  Ultimaker doesn't make definition files for other brands of printers.  They can't.  Those come from the community or from the actual manufacturer of the machine.  Your StartUp G-Code comes straight from the definition file which might be right from Wanhao.

fvrmr, 2022-01-21:
Hi @rongoodman303 thank you for your request. I will change this to a feature request. 
However I will also close this issue since we don't maintain these profiles like @GregValiant explained. Please contact the manufacturer of your machine to change this.

closed 2022-01-21 by fvrmr