Ultimaker/Cura #11290: Cura closes when trying to open files

Nprint3D, 2022-01-19:
### Application Version

4.12.1 and 4.13

### Platform

Windows 10

### Printer

Creality CR6-SE

### Reproduction steps

1) I tried to click on the file icon in the prepare window. .
2) I then tried to click file, open file. 
3) I then tried to click file, open recent.
4) Find the file in file explorer.

### Actual results

1) The open files window would briefly open and then Cura would close .
2) The open files window would again briefly open and then Cura would again close.
3) Cura would open with the file.
4) Cura would open doing this.

### Expected results

1) For the open files window to open and to select a file and Cura to open it.
2) For the open files window to open and to select a file and Cura to open it.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7894049/cura.log)
2022-01-19 05:49:51,293 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [83]: Adding QT5 plugin path: D:\Desktop\3D Printing\Ultimaker Cura 4.13.0\PyQt5\plugins
2022-01-19 05:49:51,293 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [93]: Adding QT5 plugin path: D:\Desktop\3D Printing\Ultimaker Cura 4.13.0\PyQt5\plugins
2022-01-19 05:49:51,293 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
2022-01-19 05:49:51,293 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 4.13.0
2022-01-19 05:49:51,294 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 7.9.0
2022-01-19 05:49:51,294 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type None
2022-01-19 05:49:51,300 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Desktop\3D Printing\Ultimaker Cura 4.13.0\share\uranium\plugins must be a folder.
2022-01-19 05:49:51,300 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Desktop\3D Printing\Ultimaker Cura 4.13.0\share\cura\plugins must be a folder.
2022-01-19 05:49:51,300 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Desktop\3D Printing\lib\uranium must be a folder.
2022-01-19 05:49:51,300 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Desktop\3D Printing\lib64\uranium must be a folder.
2022-01-19 05:49:51,301 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Desktop\3D Printing\lib32\uranium must be a folder.
2022-01-19 05:49:51,301 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Desktop\3D Printing\Resources\uranium\plugins must be a folder.
2022-01-19 05:49:51,301 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location D:\Desktop\3D Printing\Resources\cura\plugins must be a folder.
2022-01-19 05:49:51,301 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
2022-01-19 05:49:51,301 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is C:\Users\Nino\AppData\Roaming\cura\4.13
2022-01-19 05:49:51,301 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is C:\Users\Nino\AppData\Roaming\cura\4.13
2022-01-19 05:49:51,301 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is C:\Users\Nino\AppData\Local\cura\4.13\cache
2022-01-19 05:49:51,301 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [502]: Found config: ['C:\\Users\\Nino\\AppData\\Roaming\\cura', 'C:\\Users\\Nino\\AppData\\Local\\cura'] and data: ['C:\\Users\\Nino\\AppData\\Roaming\\cura', 'C:\\Users\\Nino\\AppData\\Local\\cura']
2022-01-19 05:49:51,301 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [506]: Latest config path: C:\Users\Nino\AppData\Roaming\cura\4.12 and latest data path: C:\Users\Nino\AppData\Roaming\cura\4.12
2022-01-19 05:49:51,301 - INFO - [MainThread] UM.Resources.copyVersionFolder [542]: Copying directory from 'C:\Users\Nino\AppData\Roaming\cura\4.12' to 'C:\Users\Nino\AppData\Roaming\cura\4.13'
2022-01-19 05:49:51,301 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: D:\Desktop\3D Printing\Ultimaker Cura 4.13.0\resources\bundled_packages\cura.json
2022-01-19 05:49:51,302 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: D:\Desktop\3D Printing\Ultimaker Cura 4.13.0\resources\bundled_packages\uranium.json
2022-01-19 05:49:51,302 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [123]: Trying OpenGL context 4.1...
2022-01-19 05:49:51,302 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [131]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
2022-01-19 05:49:51,302 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [177]: Detected most suitable OpenGL context version: 4.1 Core profile
2022-01-19 05:49:51,302 - INFO - [MainThread] UM.Qt.QtApplication.initialize [184]: Initializing job queue ...
2022-01-19 05:49:51,302 - INFO - [MainThread] UM.Qt.QtApplication.initialize [188]: Initializing version upgrade manager ...
2022-01-19 05:49:51,302 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from D:\Desktop\3D Printing\Ultimaker Cura 4.13.0\resources\bundled_packages\cura.json
2022-01-19 05:49:51,302 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from D:\Desktop\3D Printing\Ultimaker Cura 4.13.0\resources\bundled_packages\uranium.json
2022-01-19 05:49:51,302 - INFO - [MainThread] UM.PackageManager._loadManagementData [197]: Loaded user packages management file from C:\Users\Nino\AppData\Roaming\cura\4.13\packages.json
2022-01-19 05:49:51,302 - INFO - [MainThread] UM.PackageManager._saveManagementData [270]: Package management file C:\Users\Nino\AppData\Roaming\cura\4.13\packages.json was saved
[... 402 lines cut ...]
2022-01-19 06:05:18,710 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:06:18,963 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:07:20,025 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:08:20,018 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [64]: Refreshing the access token for [https://account.ultimaker.com]
2022-01-19 06:08:20,026 - DEBUG - [MainThread] UM.Logger.debug [131]: Was already busy refreshing token. Do not start a new request.
2022-01-19 06:08:20,435 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:08:20,727 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [289]: Attempting to store the auth data for [https://account.ultimaker.com]
2022-01-19 06:08:30,743 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.01598358154296875 seconds
2022-01-19 06:09:21,379 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:10:21,872 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:10:45,196 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.014949321746826172 seconds
2022-01-19 06:11:22,115 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:12:22,469 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:13:23,644 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:14:24,106 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:15:24,692 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:16:24,899 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:17:24,979 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [64]: Refreshing the access token for [https://account.ultimaker.com]
2022-01-19 06:17:24,983 - DEBUG - [MainThread] UM.Logger.debug [131]: Was already busy refreshing token. Do not start a new request.
2022-01-19 06:17:25,508 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [289]: Attempting to store the auth data for [https://account.ultimaker.com]
[... cut ...]
2022-01-19 06:17:35,529 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.020911455154418945 seconds
2022-01-19 06:18:27,259 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:19:27,466 - DEBUG - [MainThread] UM.Logger.debug [131]: Synced cloud printers with account.
2022-01-19 06:19:44,684 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.01694321632385254 seconds

fieldOfView, 2022-01-19:
If you also have Chitubox installed on the same computer, please follow this article:
https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

Nprint3D, 2022-01-19:
Perfect thank you very much for the quick response and fix. 
 
Kind Regards
Nino Coter
Nprint 3D
CELL: 082 490 8248
EMAIL: ***@***.***

 

https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

closed 2022-01-20 by nallath