Ultimaker/Cura #11292: Materials Form cannot delete new material

Gamma36, 2022-01-19:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Duplicate eSUN White PLA+ material

It cannot be deleted.

### Actual results

It cannot be deleted

### Expected results

It should be easy to remove material I create. The only workaround is to edit the material files and force a reset

[image]

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[eSUN_PLA_PRO_White+%232.xml.zip](https://github.com/Ultimaker/Cura/files/7896136/eSUN_PLA_PRO_White%2B.232.xml.zip)

[CuraFiles.zip](https://github.com/Ultimaker/Cura/files/7896221/CuraFiles.zip)

I've developed CAD/CAM C++ software for decades. The Material form is so difficult to decipher.  Looking at stderr.log, I respectfully suggest you ditch the Cloud stuff for simple home users. Perhaps a checkbox "I want to break free" :-)

Regards
Paul Hadley
BSc (Hons) Elec Eng 1981

fieldOfView, 2022-01-19:
Materials that are currently used by a printer configured in Cura cannot be deleted. There's a subtle cue that the material is the currently active material for the current printer; in the list of materials the material is italicized. "Activate" a different material and you should be able to delete this material.

> I've developed CAD/CAM C++ software for decades.

With all that experience, you are more than welcome to come contribute to Cura; it is open source after all. Help in improving it!

Gamma36, 2022-01-19:
I think best practice to ungrey Remove, then pop up a message "You cannot delete materials currently in use"

The corner stone of GUIs is to allow the novice users learn the interface.

Gamma36, 2022-01-19:
Thinking more, Activate should be replaced by Select, which is a the dominant action verb in CAD/CAM, word processing etc

fvrmr, 2022-01-25:
I have discussed it with the team and we decided to add a tool tip why you can't delete a material. 
Devs see CURA-8907

fieldOfView, 2022-01-25:
Disabled controls normally don't have a tooltip, do they?