Ultimaker/Cura #11293: Extruder error *immediately* after creating a custom material

Gamma36, 2022-01-19:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Clink on Create in the Materials form

Wait for suspicious 2 second delay

### Actual results

Immediate Extruder error on the main Cura interface

[image]

### Expected results

No errors. Custom material should load correctly for slicing and then allow me to edit temperatures

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Something has gone very wrong with Materials in 4.13.0    There is no documentation about custom temperatures either

[CuraFiles.zip](https://github.com/Ultimaker/Cura/files/7896610/CuraFiles.zip)

Gamma36, 2022-01-19:
I've had to revert to 4.12.1, to set custom temperatures. It works fine.

fvrmr, 2022-01-21:
Hi @Gamma36 thank you for your report. 
This is not an error but a warning that the material type you choose for the custom material has not a supported profile. You can still use this profile, but is not supported, which means there is not an existing material profile to set the temperatures. 
In 4.12.1 you would see this:
[image]
The difference between 4.12.1 and 4.13 is only the warning. 

Let me know if this makes sense to you.

Gamma36, 2022-01-21:
Fenne, 
It partially makes sense. But my experience as a developer makes me think the average user would not understand "existing material profile". 

The materials database needs to be normalised and the form rewritten. Also Assign is rarely used and confusing (assign implies 2 arguments), Select is a commonly understood verb and that's what the button does.

fvrmr, 2022-02-11:
We have a ticket on our backlog for when creating a custom material to have a dropdown menu to choose what material you have. Because it seems like there is the problem, if you type PLA R it will show that warning icon and show not supported, because PLA R is an unknown material type to match with your profiles. See https://github.com/Ultimaker/Cura/issues/8229
I will close this issue and mark it as a duplicate, if you have remarks please post them in the other thread so we keep everything in one issue.

closed 2022-02-11 by fvrmr