Ultimaker/Cura #11294: Ultimaker Cura CRASHING in every version.

jkonttila, 2022-01-19:
### Application Version

ALL VERSIONS

### Platform

Windows 10 Pro

### Printer

Ender 3 Pro Max

### Reproduction steps

1. Open Cura slicer
2. Cura slicer opens and shuts down without any error messages.
3. Compatibility mode doesn't help. Cura still crashing.
4. Checked all drivers, reinstalled software. Tested all versions possible. Nothing works.
5. Checked GPU settings, tested everything. Reset all. Tried everything that is possible to try in AMD settings. Nothing works.
6. Send multiple error reports to AMD, Windows and Ultimaker. 0 Responses in couple months.
7. Error has been going on a full year now. Slicer crashing and nothing seems to fix it. Ultimaker customer service cant help. 0 responses on error reports. Frustration, nerve loss, faith in total Ultimaker service gone.
8. When I try to open the Cura several times, the GPU freezes and the screen goes blank for a while. This only happens in the Cura slicer.

### Actual results

Nothing happens. nothing seems to fix this problem. I think that Cura slicer has some major issues with AMD gpu's. Im using AMD Radeon RX580.

### Expected results

I want these software's to work like they should.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7896631/cura.log)
[atikmdag_dce.log](https://github.com/Ultimaker/Cura/files/7896636/atikmdag_dce.log)

Cura log file and also my AMD Radeon RX580 GPU log file.

nallath, 2022-01-20:
I can see in your logs that you have sliced a few models, so it seems that it doesn't crash all the time. It might be related to some issues with Ryzen / AMD cards, you can find some more info about that here https://support.ultimaker.com/hc/en-us/articles/360018633500-Ultimaker-Cura-won-t-run-on-my-Ryzen-Graphics-card and https://github.com/Ultimaker/Cura/issues/5265

On a final note, I hope you also realize that Ultimaker only provides support for products that we sell. Cura is provided free of charge. Although we try our best to help everyone that uses Cura, we will give priority to users of our machines. If you want support for your printer, it's best to contact Creality. They also provide a version of Cura that they maintain and develop.

fieldOfView, 2022-01-25:
> It might be related to some issues with Ryzen / AMD cards

PSST... Ryzen refers to a CPU architecture, not a GPU. There is no such thing as a Ryzen Graphics card. That article does not really inspire confidence that Ultimaker knows what they are talking about.

Ghostkeeper, 2022-01-27:
Yeah that article is a bit wrong. I asked the people responsible to make corrections. I made a summary of the issue here that might be a better source: https://github.com/Ultimaker/Cura/issues/5265#issuecomment-460158267

However the symptoms there are a bit different. The symptom of that issue is that the screen goes black and nothing on the computer responds any more. That is different from the "shuts down without any error messages" symptom, and it's not just going black "for a while".

closed 2022-02-04 by no-response[bot]

Charles-Isaac, 2022-09-27:
I have the same hardware configuration and cura will crash 90% of the time when I try to view the preview.