Ultimaker/Cura #11295: Support blockers not replicated correctly when object is multiplicated

bleoncz, 2022-01-19:
### Application Version

4.12.1

### Platform

MacOS

### Printer

Creality 10s

### Reproduction steps

1. Opened single object (from original STL file - own design)
2. Added custom, Cura-native support blockers to some locations
3. Duplicated object (with Cura-native multiplication feature)
4. Support blockers in cloned object appear as additional material, not as support blockers
5. Re-started OS
6. Re-started Cura
7. Re-opened single object from STL file
8. Re-duplicated it
9. Same issue persists

### Actual results

issue persists

### Expected results

Duplicated object should retain the same settings as the original one - here: custom support blockers at exact same location as the original.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

there is no additional information.

fvrmr, 2022-01-21:
Hi @bleoncz thank you for your report.
I can reproduce your issue in 4.13 not in 4.12.1. I will bring this up for discussion, keep you posted!

fvrmr, 2022-02-01:
Hi @bleoncz I could not reproduce this issue anymore in 4.13.1. Could you check for me, if you can still reproduce this?
If so, does it happen with one file or multiple?

bleoncz, 2022-02-01:
it was 4.12.1 i have referred to.

fvrmr, 2022-02-01:
I could not reproduce this in 4.12.1 at all. Does it still happen in 4.12.1 with every file?

bleoncz, 2022-02-01:
i haven’t tried with other files yet, will check

closed 2022-02-17 by no-response[bot]