Ultimaker/Cura #11296: TEVO BLACK WINDOW в CURA не отображаются материалы

inrudjt, 2022-01-19:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Tevo Black Widow

### Reproduction steps

Добрый день при добавлении других принтеров в них можно выбрать материал и увидеть затраты пластика при печати, На данной модели принтера нет возможности выбора материала, а также добавление нового. Прошу решить проблему скрины прилогаю 
[image]
 

### Actual results

не нашел что возможно сделать в данной ситуации

### Expected results

Вот как выглядит на других моделях принтера 

[image]
[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

прошу подсказать решение проблемы

fieldOfView, 2022-01-20:
Please use english in this issue queue. If we cannot read what you are saying, we cannot help you.

For many of the people helping others in this issue queue, english is not their native language either. We use english as a common ground. Many of us put in our own time to help you. Please do not also ask us to use a translator to help you.

inrudjt, 2022-01-20:
> Кто бы ни составлял файл определения вашего принтера («Тобби»), он не хотел, чтобы в нем были материалы. "metadata": { "visible": true, "author": "TheTobby", "manufacturer": "Tevo", "file_formats": "text/x-gcode", **"has_materials": false** , поскольку установлено значение False, вам также не разрешено импортировать материалы. Может быть, Tevo может подсказать, почему они это сделали и как это обойти.

Это только в данной модели принтера от TEVO в других от компании TEVO все впорядке

fvrmr, 2022-01-21:
Closing this issue, since we don't maintain the printer definition of third party printers. 
Please contact your printer manufacturer for this.

closed 2022-01-21 by fvrmr