Ultimaker/Cura #11297: longer

emanuel1979, 2022-01-20:
### Application Version

4.9

### Platform

windows

### Printer

longer

### Reproduction steps

Cura doesn't have longer with the .6 and .8 nozzles. I think its not even listed this time. Please help me.

### Actual results

Cura doesn't have longer with the .6 and .8 nozzles. I think its not even listed this time.

### Expected results

Please add the longer printer with all the nozzles. Me EMAIL is emanuel.calderon@pettorino.com

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Cura doesn't have longer with the .6 and .8 nozzles. I think its not even listed this time.

emanuel1979, 2022-01-20:
Cura doesn't have longer with the .6 and .8 nozzles. I think its not even listed this time.

GregValiant, 2022-01-20:
That was up to whoever wrote the definition file (it looks like Longer wrote it).  I think you need to go back to Longer and ask them to add more nozzles or you can add your own.

fvrmr, 2022-01-21:
Hi @emanuel1979 thank you for your report.
We don't maintain these profiles, so we can't change this.
Like @GregValiant mentioned please contact the manufacturer or contributor to add this.  

I will close this issue.

closed 2022-01-21 by fvrmr

emanuel1979, 2022-01-22:
Thanks. I which I could change it my self. Its simple. Thanks for the advice .

 Sign up for FREE email from Everyone.net email at http://www.everyone.net