Ultimaker/Cura #11298: Add new Geeetech printers to the printer list. Thank you!

Geeetech3D, 2022-01-20:
### Is your feature request related to a problem?

Dear Sir or Madam,
Hi, I'm an engineer at Geeetech and I've been using Cura as my main slicing software. It's very useful and I'm always appreciate it.
It's just that we have some new 3D printers that can't be found in your printer list, such as Mizar, Mizar S, Mizar Pro, Mizar M, Mizar T, etc. Can you please add them.

Feel free to contact me if you have any questions.

### Describe the solution you'd like

We have some setup suggestions for it and some relevant curaprofile, hope can add in your "Add a non-networked printer" list of the Cura slicing software.

### Describe alternatives you've considered

Are there any setup sheets for us to fill out and for your reference or give us some suggestion, so that our customers can use Cura more easily. Thank you in advance.

### Affected users and/or printers

Geeetech customers.

### Additional information & file uploads

[https://drive.google.com/file/d/1Lo5o8_Ns_iaqx7l8NkqlAQt7eXA0qQCb/view?usp=sharing](url)
Well-setting curaprofile for reference.

GregValiant, 2022-01-20:
All third party printer definitions come from the manufacturer (you) or from the community.  Ultimaker writes the ones for Ultimaker printers.  I don't know the proper procedure for submitting them to Ultimaker.  It may be as simple as submitting them via a pull request here.

fvrmr, 2022-01-21:
Hi @Geeetech3D thank you for reaching out. 
You can submit these profiles via a pull request on the master branch of the `Ultimaker/Cura` repository. 
I will close this issue, but let me know if you have questions or need help by creating a pull request.

closed 2022-01-21 by fvrmr

Geeetech3D, 2022-01-22:
Hi, 
Thank you for your concerning. I will do that. 
At the same time, do you have email address for me to contact you in private? Or you can contact me via my email: barry.ding<barry.ding@geeetech.cn>. Thank you!

Best regards

fvrmr, 2022-01-25:
I will send you an email!