Ultimaker/Cura #11299: Option to pause between printing multiple objects 'one at a time'

kylane, 2022-01-20:
### Is your feature request related to a problem?

Kinda.. Sometimes when printing large objects (imagine a tower) that is higher than the X gantry, it can interfere with the next print, and even jam the gantry when Z is returning to 0 to start the next print. Can actually cause damage to the printer.

Perhaps, if the previous print is higher than the bottom of the gantry when Z = 0, then do the beep/pause? Or the option... "Only pause on possibly problematic prints/all prints"

### Describe the solution you'd like

Would be nice for a beep and a pause perhaps to allow the user to move the completed object before continuing.

### Describe alternatives you've considered

Ive tried adding pauses "per layer" and trying to get it to work, but I cant seem to pause it after the top later and before the new first layer on the next object.

### Affected users and/or printers

Can imagine it would affect most printers, cartesian or delta given any particular layout of objects?

fvrmr, 2022-01-25:
Hi @kylane thank you for your feature request. 
Did you use the post processing script pause at height of at layer? Because I would assume that both of them should work.

kylane, 2022-01-27:
> Hi @kylane thank you for your feature request. Did you use the post processing script pause at height of at layer? Because I would assume that both of them should work.

Hi there - sure did. I could get it to stop before printing the last layer of the first object, or at the start on the 2nd object (by then the gantry had already moved and potentially caused an issue).

But I may have got it wrong or something - it was a while ago I last tried. But a checkbox to tick, and just have the printer head move away, allow us to remove the first print, then continue would be a better user experience in my humble opinion :)

Ghostkeeper, 2022-01-27:
It's possible to add post-processing scripts to pause after every object, although the pause condition should be a layer # then, not a certain height (because that height gets reached multiple times and they'll all trigger the first time that height is reached). However I think the feature request here is a bit more than that. The changes I see here are:

* Remove the gantry height limit when printing in one-at-a-time mode.
* After printing something that is higher than the gantry height, pause the print.
* Unless it is the last object.
* Make it clear to the user that while they can exceed the gantry height, it causes the printer to pause.
* Configure the way in which to pause similar to the post-processing script (because different firmware needs different pause commands).

An alternative for this would be to have a printer with a print queue, like what OctoPrint has or the Ultimaker firmware. For this reason though, I don't think Ultimaker will help with developing this. Ultimaker users can just slice a couple of things and send each of them to the queue, to start one after another after removing the previous print.

closed 2023-07-13 by github-actions[bot]