Ultimaker/Cura #11300: Crash on drag&drop

sladaar, 2022-01-20:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Anycubic i3 mega

### Reproduction steps

1. Drag & drop to open file

### Actual results

Cura crashes!

### Expected results

Opening the file...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

2022-01-20 11:33:29,862 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.011558771133422852 seconds
2022-01-20 11:35:29,679 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1770]: Attempting to read file file:///D:/Anycubic/Food+Chain+Magnate+Organizer/files/FCM_Marketing-1.stl
2022-01-20 11:35:29,682 - INFO - [JobQueueWorker [4]] STLReader.STLReader.load_file [53]: Using NumPy-STL to load STL data.
2022-01-20 11:35:29,688 - DEBUG - [JobQueueWorker [4]] UM.Mesh.MeshData.calculateNormalsFromVertices [558]: Calculating normals took 0.0 seconds
2022-01-20 11:35:29,689 - DEBUG - [JobQueueWorker [4]] STLReader.STLReader._read [105]: Loaded a mesh with 6192 vertices
2022-01-20 11:35:29,699 - DEBUG - [JobQueueWorker [4]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.005001544952392578 seconds. 27 input vertices. 27 output vertices.
2022-01-20 11:35:29,894 - DEBUG - [JobQueueWorker [4]] UM.FileHandler.ReadFileJob.run [83]: Loading file took 0.2 seconds
2022-01-20 11:35:29,904 - DEBUG - [JobQueueWorker [4]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.005028486251831055 seconds. 27 input vertices. 27 output vertices.
2022-01-20 11:35:29,916 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x000001B6E6AAA520>, took 2ms
2022-01-20 11:35:29,937 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=2) RotateOp.(node=<CuraSceneNode object: 'FCM_Marketing-1.stl'>,rot.=Q<0.0,0.0,0.0,w=4.0>) TranslateOp.(node=<CuraSceneNode object: 'FCM_Marketing-1.stl'>,trans.=<0.000,0.000,0.000>), took 2ms
2022-01-20 11:35:30,142 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [424]: file:///C:/Program Files/Ultimaker Cura 4.13.0/resources/qml/ObjectItemButton.qml:79:9: QML Button: Binding loop detected for property "width"
2022-01-20 11:35:30,146 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [424]: file:///C:/Program Files/Ultimaker Cura 4.13.0/resources/qml/ObjectItemButton.qml:79:9: QML Button: Binding loop detected for property "width"
2022-01-20 11:35:30,150 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [424]: file:///C:/Program Files/Ultimaker Cura 4.13.0/resources/qml/ObjectItemButton.qml:79:9: QML Button: Binding loop detected for property "width"

fieldOfView, 2022-01-20:
We need the entire log file to diagnose this. Nothing in the snippet you provided says anything about a crash.

Does Cura also crash when you use File -> Open... instead of drag and drop?

sladaar, 2022-01-20:
[cura.log](https://github.com/Ultimaker/Cura/files/7904437/cura.log)

Sorry, here it is!

And no, it doesn't. It just crashes on drag&drop!

fvrmr, 2022-01-25:
Hi @sladaar thank you for your log.
Unfortunately your log doesn't give any information why it is crashing. I could also not reproduce this issue myself. 
Does it only happen in 4.13?

closed 2022-02-08 by no-response[bot]