Ultimaker/Cura #11302: Can't upload to Arduino nano 33 IOT while cura is running

mbueno0, 2022-01-20:
### Application Version

4.11.0

### Platform

Windows 11 pro (brand new installation as of this week)

### Printer

Prusa i3 mk3s and/or Creality cr-10s (seemingly not printer-specific)

### Reproduction steps

**I'm having issues uploading to an Arduino nano 33 IOT while Cura is running.**

When Cura is running, I get complaints from the Arduino IDE (**version 1.8.16**) that the COM port to which I'm trying to upload my Arduino sketch is unavailable. When Cura is not running, the uploads happen just fine.  I determined that Cura was the culprit by following the instructions given here:  [https://michaelbergeron.com/blog/arduino-com-port-busy](https://michaelbergeron.com/blog/arduino-com-port-busy).

This does not happen with AVR-based Arduino boards (nano, uno, mega, etc.), just the SAMD-based nano 33 IOT I have.

Running Microsoft's Process Explorer according to the instructions given in the link above was instrumental in identifying the process responsible. In Process Explorer, I did '**find file handle or DLL**' and entered '**usbser**' as the parameter. The result said '**\Device\USBSER001**' was in use by cura.exe.

I tried all the different options for changing the default behavior for when removable media is inserted, thinking that might help. It did not.

**These are the error messages I'm getting from the Arduino IDE:**

> `Sketch uses 11768 bytes (4%) of program storage space. Maximum is 262144 bytes.
> Global variables use 3568 bytes (10%) of dynamic memory, leaving 29200 bytes for local variables. Maximum is 32768 bytes.
> processing.app.debug.RunnerException
> 	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:152)
> 	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
> 	at processing.app.SketchController.upload(SketchController.java:732)
> 	at processing.app.SketchController.exportApplet(SketchController.java:703)
> 	at processing.app.Editor$UploadHandler.run(Editor.java:2081)
> 	at java.lang.Thread.run(Thread.java:748)
> Caused by: processing.app.SerialException: Error touching serial port 'COM4'.
> 	at processing.app.Serial.touchForCDCReset(Serial.java:107)
> 	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:136)
> 	... 5 more
> Caused by: jssc.SerialPortException: Port name - COM4; Method name - openPort(); Exception type - Port busy.
> 	at jssc.SerialPort.openPort(SerialPort.java:164)
> 	at processing.app.Serial.touchForCDCReset(Serial.java:101)
> 	... 6 more
> `

I realize this could be related to many 3D printers having Arduinos inside them. This is a problem for me because I'm **always** running (typically multiple instances of) Cura. The issue I'm having was while running a single instance, however.

Thanks for any ideas/suggestions! :)

### Actual results

It is all in the first step

### Expected results

It is all in the first step

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

For what it is worth, here is my Cura log file:

[cura.log](https://github.com/Ultimaker/Cura/files/7906210/cura.log)

fieldOfView, 2022-01-20:
If you don't use USB printing in Cura, you can disable the USB Printing plugin via the "Installed" tab of the Marketplace. That will stop this behavior.

fvrmr, 2022-01-21:
Read here how to disable the USB printing plugin: https://support.ultimaker.com/hc/en-us/articles/4407742254610-Ultimaker-Cura-interacts-with-microcontrollers

Closing this issue, since that will probably solve your issue, if not feel free to reach out!

closed 2022-01-21 by fvrmr

mbueno0, 2022-01-21:
Thanks all! This did solve my problem!
Have a fantabulous day!

<https://chrome.google.com/webstore/detail/bitmoji/bfgdeiadkckfbkeigkoncpdieiiefpig>
Sincerely,
     Michael Bueno

*"To invent, you need a good imagination and a pile of junk." *
*--**Thomas A. Edison*

<https://chrome.google.com/webstore/detail/bitmoji/bfgdeiadkckfbkeigkoncpdieiiefpig>