Ultimaker/Cura #11304: Änderungen der Drucktemperatur werden nicht übernommen

Bodo62, 2022-01-20:
### Application Version

4.13.0

### Platform

Win 10

### Printer

Anycubic I3 mega S

### Reproduction steps

Bauteil geladen.
Profil in den Druckeinstellungen gewählt
Bei Material die Drucktemperatur auf 220 Grad C gestellt
Profil mit aktuellen Druckeinstellungen überschrieben
Slice ausgeführt
Auf Festplatte speichern
Ordner öffnen
Datei auf SD Karte geschoben
Datei im Drucker geladen

### Actual results

Drucktemperatur beträgt immer nur 200 Grad C

### Expected results

Drucktemperatur von 220 Grad C soll eingestellt werden

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7907348/cura.log)

[AI3M_Hebel.zip](https://github.com/Ultimaker/Cura/files/7907391/AI3M_Hebel.zip)

GregValiant, 2022-01-20:
You have the Printing Temperature Initial Layer set to 200 or the Initial Printing Temperature set to 200.

Just above all the Cura settings is the "Settings Search" box.  Just to the right of that is an icon with three lines on it.  Click on that icon and set the visibility to "All" and then go down to the Material section and look at the temperatures.

Ghostkeeper, 2022-01-27:
Please report issues in English. Not everyone can read German, but most internet users can read English.

Bodo62, 2022-02-01:
Hello and thanks for the quick support.
I do not understand why all values from the configuration "ALL" are taken
into account, although I have not checked any of the boxes in the custom
interface.

Ghostkeeper, 2022-02-15:
All settings are used when slicing. Whether or not a setting gets used is not dependent on whether they are made visible by the user in their user interface.

Profiles can override settings that are not made visible, too. For instance, a particular material profile may specify that the initial layer temperature should always be 200°C regardless of what the printing temperature is. Or it could say that it's the printing temperature+10°C. In your case, it's most likely the former. But it's impossible to know without seeing a project file.

GregValiant, 2024-11-16:
I'll remove the bug label and close this.

closed 2024-11-16 by GregValiant