Ultimaker/Cura #11305: Upgrade Option

Cthulhu709, 2022-01-21:
### Is your feature request related to a problem?

Starting with v4.8.0 for me, and likely earlier, each version of Cura installs, do not upgrade existing software.

This causes issues with profiles for machines and print settings.

### Describe the solution you'd like

I'd like to see an option to install an upgrade, so each version carries persistent settings and preferences that I've become accustomed to.

### Describe alternatives you've considered

I have performed the alternative, which is to manually implement the preferences for print profiles, which I use to fine-tune my printer for each filament I purchase.  Every brand and product I've used, I perform a series of initial prints, which allow me to fine-tune the filament attitude out, giving me exceptional prints without fighting with it.  I currently have approximately 35 profiles, making updating my version of Cura, very time-consuming, instead of a simple process.

### Affected users and/or printers

All your users who are dragging their heels updating, because of the lack of continuity of their configuration.

Cthulhu709, 2022-01-21:
Never mind... just installed the new version to test drive it... and my profiles are copied over to the new version.. 

Bravo git team.. Bravo!

fvrmr, 2022-01-21:
Hi @Cthulhu709 thank you for your request. 
You can install multiple Cura versions on one computer. To upgrade your settings to the next version you can also create a backup.

Closing this issue since, but feel free to reach out when needed!

closed 2022-01-21 by fvrmr