Ultimaker/Cura #11306: Support roof doesn't cover all support top

maaboo, 2022-01-21:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Put model required supports into slicer
2. Enable tree support and slice

### Actual results

Roof only partially covers support top

### Expected results

Roof covers all support top

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[roofbug.zip](https://github.com/Ultimaker/Cura/files/7911417/roofbug.zip)

fvrmr, 2022-01-21:
Hi @maaboo thank you for your report.
I could reproduce your issue in 4.12, but not in 4.13. 
However in 4.13 there is a known bug with tree support you might run into when using it. See: https://github.com/Ultimaker/Cura/issues/11258

maaboo, 2022-01-23:
I confirm that this particular bug is no longer reproducible in 4.13 and I also have [#11258](https://github.com/Ultimaker/Cura/issues/11258). Also I noticed that top of the support roof looks "dirty" like it's density significantly decreased. Is this a bug or just configuration? I thought interface should be consistent on all the layers.

fvrmr, 2022-01-25:
Could you show me with a screenshot what you mean?

maaboo, 2022-01-26:
[image]
Looks more like rice

fvrmr, 2022-01-27:
@maaboo thank you, could you share a project file of this? Because with your other project file I could not reproduce this.

maaboo, 2022-01-27:
This is strange, but here you are:
[roofbug2.zip](https://github.com/Ultimaker/Cura/files/7950610/roofbug2.zip)

fvrmr, 2022-01-28:
I have discussed it with the team and this is expected behaviour because of the support branch diameter. If you change this to 0.6mm it will show the interface on every layer the same!

maaboo, 2022-01-28:
Well, not branch diameter but branch distance. I changed it as you suggested and it works as expected.
I think there is nothing left and issue could be closed.
Thanks.

closed 2022-01-28 by maaboo