Ultimaker/Cura #11310: Print Sequence, "One at a Time" optimization / improvements

UberEclectic, 2022-01-21:
### Is your feature request related to a problem?

Print sequence, "one at a time" doesn't work with more than one extruder and in general part placement does a poor job of dealing with gantry avoidance.  Issues are complicated by no obvious means to bias the order in which parts are printed.

### Describe the solution you'd like

Because of the different types of printers, gantry, and head orientation it would be helpful to capture more information about the type of gantry and the size of the print heads and use that information during part placement to avoid object contact.
A human being can work out where to place parts and what sequence to print them in to avoid gantry and/or print head coming in contact with parts, so it's perfectly feasible for software to figure out the optimal part placement and printing sequence.

### Describe alternatives you've considered

I've always had to lie about the gantry height and then manually place parts.
One problem is if the order the parts is printed isn't "safe" for gantry clearance then I may have to swap parts.
CURA seems to decide which object to print first by a scheme unknown to me, but thankfully it doesn't change which part is first once I move that part to another place on the build plate.

### Affected users and/or printers

Any/all.  But especially dual head like Lulzbot TAZ PRO, and Craftbot FLOW IDEX (XL).
In the case of the Craftbot IDEX, the print heads are mounted quite some distance in front of the gantry, so the optimal print sequence would be to print parts near the front of the bed and sequence parts from front to back of the bed.

### Additional information & file uploads

Further background on this use case...

I like to use "One at a Time" print sequence, when printing multiple objects in the same job.
It reduces travel time and avoids "hairs" spanning between parts.
There are a few issues.

1. CURA does not automatically figure out where to place objects to avoid the gantry toppling them over.  It only works if the objects are less than the gantry height.  My usual workaround is to temporarily lie about the gantry height and then manually place the objects to avoid the gantry.  This should be a simple thing for CURA to figure out, at least in the case of exactly two parts -- just place one part toward the front and the other part toward the back with a gantry-width amount of space between the parts (which is what I do manually).  Likely an RFE here that the printer definition needs to capture the X/Y size of the gantry and/or print-head, not just the gantry Z-height.
2. CURA 4.x blocks the use of "One at a Time" if extruder count is >1.  I can't see the reason for this.  I used to use "One at a Time" with great success with two extruders in CURA 3.x (caveat being #1 above).   This restrictions means I can't print two color parts, nor can I print a bunch of different one-color parts, from two different extruders.
3. I got a new IDEX printer and the two extruders are mounted on the front of the gantry, quite some distance forward.  This leads to a preference to print parts at the front of bed first, then at the rear.  I don't see a way to specify the print-order for the objects.  My workaround is to see what CURA intends to do, then swap the parts if necessary.

fvrmr, 2022-01-28:
Hi @UberEclectic thank you for your request. 
This request would only be beneficial for IDEX printers. Ultimaker printers are not IDEX printers, so this would get no priority for us to implement the option to print with two extruders in one at a time mode. 
If you want more information how to measure the print head settings and gantry height, you can download the settings guide in the marketplace: https://marketplace.ultimaker.com/app/cura/plugins/Ghostkeeper/SettingsGuide2
[image]

So based on that the I have to reject this request. I hope you understand.

closed 2022-01-28 by fvrmr

Ghostkeeper, 2022-01-28:
The order is determined automatically when a slice starts, but only for cases where the order would be a problem according to the print head shape that Cura knows. If Cura knows no print head shape, there are no collisions and it will indeed use the order in the file list as a tie breaker.

strongly-typed, 2023-06-05:
> This request would only be beneficial for IDEX printers. Ultimaker printers are not IDEX printers, so this would get no priority for us to implement the option to print with two extruders in one at a time mode.

We use the dual extruder of several S5 and S3 to mix materials, e.g. PVA and PLA or two colours of PLA for coloured logos in parts.
For small parts (small relative to the build surface) it totally would make sense for us to print parts "one at a time" as the surface quality is improved.