Ultimaker/Cura #11311: Support blocker in 4.13.0

ReaperKeeper, 2022-01-21:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Creality CR-10S

### Reproduction steps

1. Loaded a helmet file
2. Added support blocker to bottom dome of helmet
3. Scaled to size
4. Sliced
5. Noticed print was over 1700G of filament and questioned why
6. Went to preview, noticed in the line view that it printed supports anyway.

### Actual results

Cura does not recognize the support blockers and is creating supports where asked not to.

### Expected results

Should block supports

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Great Ape Studios Mandalorian helmet - can't share file due to copyright.
[image]
[image]

fvrmr, 2022-01-24:
Hi @ReaperKeeper thank you for your report.
What are exactly the steps you are doing? You add one support blocker and then resize this? 
Or do you want to duplicate your support blockers?

Does this also happen on another model?

ReaperKeeper, 2022-01-24:
One support blocker scaled up. I went back to 4.11 and it worked.

It did the same for other models. Adding individuals didn't help either.

fvrmr, 2022-01-27:
Could you share a project file (with another model) where it is also happens?

closed 2022-02-10 by no-response[bot]