Ultimaker/Cura #11312: Issue

DevooUP96, 2022-01-21:
### Application Version

4.13

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Reinstall and still crash when i try to save sliced file aaaaaaaaaa

### Actual results

Still persit 

### Expected results

Fix the issue 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

2022-01-21 21:34:03,331 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [83]: Adding QT5 plugin path: C:\Program Files\Ultimaker Cura 4.13.0\PyQt5\plugins
2022-01-21 21:34:03,331 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [93]: Adding QT5 plugin path: C:\Program Files\Ultimaker Cura 4.13.0\PyQt5\plugins
2022-01-21 21:34:03,331 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
2022-01-21 21:34:03,331 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 4.13.0
2022-01-21 21:34:03,331 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 7.9.0
2022-01-21 21:34:03,331 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type None
2022-01-21 21:34:03,331 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location C:\Program Files\Ultimaker Cura 4.13.0\share\uranium\plugins must be a folder.
2022-01-21 21:34:03,331 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location C:\Program Files\Ultimaker Cura 4.13.0\share\cura\plugins must be a folder.
2022-01-21 21:34:03,331 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location C:\Program Files\lib\uranium must be a folder.
2022-01-21 21:34:03,331 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location C:\Program Files\lib64\uranium must be a folder.
2022-01-21 21:34:03,332 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location C:\Program Files\lib32\uranium must be a folder.
2022-01-21 21:34:03,332 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location C:\Program Files\Resources\uranium\plugins must be a folder.
2022-01-21 21:34:03,332 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location C:\Program Files\Resources\cura\plugins must be a folder.
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is C:\Users\FreeX\AppData\Roaming\cura\4.13
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is C:\Users\FreeX\AppData\Roaming\cura\4.13
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is C:\Users\FreeX\AppData\Local\cura\4.13\cache
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [502]: Found config: ['C:\\Users\\FreeX\\AppData\\Roaming\\cura', 'C:\\Users\\FreeX\\AppData\\Local\\cura'] and data: ['C:\\Users\\FreeX\\AppData\\Roaming\\cura', 'C:\\Users\\FreeX\\AppData\\Local\\cura']
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [506]: Latest config path: C:\Users\FreeX\AppData\Roaming\cura\4.6 and latest data path: C:\Users\FreeX\AppData\Roaming\cura\4.6
2022-01-21 21:34:03,332 - INFO - [MainThread] UM.Resources.copyVersionFolder [542]: Copying directory from 'C:\Users\FreeX\AppData\Roaming\cura\4.6' to 'C:\Users\FreeX\AppData\Roaming\cura\4.13'
2022-01-21 21:34:03,332 - WARNING - [MainThread] UM.Preferences._loadFile [192]: Old config file found, ignoring
2022-01-21 21:34:03,332 - WARNING - [MainThread] UM.Preferences.__initializeSettings [124]: Read the preferences file before initializing settings!
2022-01-21 21:34:03,332 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 4.13.0\resources\bundled_packages\cura.json
2022-01-21 21:34:03,332 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 4.13.0\resources\bundled_packages\uranium.json
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [123]: Trying OpenGL context 4.1...
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [131]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
2022-01-21 21:34:03,332 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [177]: Detected most suitable OpenGL context version: 4.1 Core profile
2022-01-21 21:34:03,332 - INFO - [MainThread] UM.Qt.QtApplication.initialize [184]: Initializing job queue ...
2022-01-21 21:34:03,333 - INFO - [MainThread] UM.Qt.QtApplication.initialize [188]: Initializing version upgrade manager ...
2022-01-21 21:34:03,333 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 4.13.0\resources\bundled_packages\cura.json
2022-01-21 21:34:03,333 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 4.13.0\resources\bundled_packages\uranium.json
2022-01-21 21:34:03,333 - INFO - [MainThread] UM.PackageManager._loadManagementData [197]: Loaded user packages management file from C:\Users\FreeX\AppData\Roaming\cura\4.13\packages.json
2022-01-21 21:34:03,333 - INFO - [MainThread] UM.PackageManager._saveManagementData [270]: Package management file C:\Users\FreeX\AppData\Roaming\cura\4.13\packages.json was saved
2022-01-21 21:34:03,333 - INFO - [MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [119]: Loading plugin configuration file 'C:\Users\FreeX\AppData\Roaming\cura\4.13\plugins.json'
2022-01-21 21:34:03,333 - WARNING - [MainThread] UM.Logger.warning [143]: Plugin location C:\Program Files\lib\cura must be a folder.
2022-01-21 21:34:03,336 - INFO - [MainThread] UM.Logger.info [137]: Loaded plugin FileLogger
2022-01-21 21:34:03,362 - INFO - [MainThread] UM.Logger.info [137]: Loaded plugin FirmwareUpdateChecker
2022-01-21 21:34:03,381 - INFO - [MainThread] UM.Logger.info [137]: Loaded plugin FirmwareUpdater
2022-01-21 21:34:03,398 - INFO - [MainThread] UM.Logger.info [137]: Loaded plugin GCodeGzReader
[... 710 lines cut ...]
2022-01-21 22:32:22,638 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2022-01-21 22:32:22,640 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2022-01-21 22:32:22,640 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend finished slicing. Resetting process and socket.
2022-01-21 22:32:24,139 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PreviewStage
2022-01-21 22:32:24,141 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [424]: file:///C:/Program Files/Ultimaker Cura 4.13.0/resources/qml/PrintSetupSelector/Custom/QualitiesWithIntentMenu.qml:168:25: QML Binding: Not restoring previous value because restoreMode has not been set.
This behavior is deprecated.
You have to import QtQml 2.15 after any QtQuick imports and set
the restoreMode of the binding to fix this warning.
In Qt < 6.0 the default is Binding.RestoreBinding.
In Qt >= 6.0 the default is Binding.RestoreBindingOrValue.

2022-01-21 22:32:24,173 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SimulationView
2022-01-21 22:32:24,174 - INFO - [MainThread] STLReader.STLReader.load_file [53]: Using NumPy-STL to load STL data.
2022-01-21 22:32:24,184 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [558]: Calculating normals took 0.000499725341796875 seconds
2022-01-21 22:32:24,185 - DEBUG - [MainThread] STLReader.STLReader._read [105]: Loaded a mesh with 12612 vertices
2022-01-21 22:32:24,188 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [60]: Loading [C:\Program Files\Ultimaker Cura 4.13.0\plugins\SimulationView\simulationview_composite.shader]...
2022-01-21 22:32:24,219 - DEBUG - [JobQueueWorker [4]] CuraEngineBackend.ProcessSlicedLayersJob.run [74]: Processing new layer for build plate 0...
2022-01-21 22:32:24,388 - DEBUG - [JobQueueWorker [4]] CuraEngineBackend.ProcessSlicedLayersJob.run [254]: Processing layers took 0.16752910614013672 seconds
[... cut ...]
2022-01-21 22:32:24,465 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [60]: Loading [C:\Program Files\Ultimaker Cura 4.13.0\resources\shaders\toolhandle.shader]...
2022-01-21 22:32:24,468 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [60]: Loading [C:\Program Files\Ultimaker Cura 4.13.0\resources\shaders\color.shader]...
2022-01-21 22:32:24,471 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [60]: Loading [C:\Program Files\Ultimaker Cura 4.13.0\resources\shaders\striped.shader]...
2022-01-21 22:32:24,500 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onProcessLayersFinished [996]: See if there is more to slice(2)...

BigBird0000, 2022-01-23:
I have a similar problem.... yesterday 4.12 would stop responding, so I tried downloading 4.13. Neither one responds, Cura starts up, but as soon as I click any menu item, cura stops responding. I have the same warnings in my cura.log file as above, which is how I found this page.

Ghostkeeper, 2022-01-28:
Do you have any shell extensions installed on Windows? For instance, the Chitubox thumbnail generator is known to cause Cura to crash when opening a file dialogue (see [this support article](https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file)).

The reproduction steps in this ticket in general don't make sense. Re-installing Cura is not expected to fix any crashes unless the Program Files folder for Cura was edited after installation (which is not in your reproduction steps). Think about how *someone else than you* could go about triggering the crash you're seeing. I'm supposing it happens when saving a file, right? How are you saving that file? Saving the g-code to disk, perhaps? Or are you saving to a USB stick?

I think @BigBird0000 has different symptoms, since they indicate a freeze instead of a crash and that it happens when opening a menu item instead of when saving a file.

closed 2022-02-14 by no-response[bot]