Ultimaker/Cura #11313: Crash on startup. empty material

chernyshof, 2022-01-22:
### Application Version

4.13.0

### Platform

arch linux

### Printer

ender 3 pro

### Reproduction steps

1. Started cura

### Actual results

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/cura/Settings/ExtruderManager.py", line 425, in getExtruderHasQualityForMaterial
    active_material_node = active_variant_node.materials[extruder_stack.material.getMetaDataEntry("base_file")]
KeyError: 'empty_material'

### Expected results

Working cura
At least send report feature it says: "SentrySDK is not available and the report could not be sent."

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Thread 0x00007f27677fe640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.10/threading.py", line 946 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007f2767fff640 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.10/threading.py", line 946 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007f27764fc640 (most recent call first):
  File "/usr/lib/python3.10/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/usr/lib/python3.10/threading.py", line 946 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007f2776cfd640 (most recent call first):
  File "/usr/lib/python3.10/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/usr/lib/python3.10/threading.py", line 946 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007f27b4ff9640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007f27b57fa640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
[... 132 lines cut ...]

Thread 0x00007f2801044640 (most recent call first):
  File "/usr/lib/python3.10/site-packages/cura/Machines/ContainerTree.py", line 181 in run
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 133 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007f2801845640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Current thread 0x00007f280ee89740 (most recent call first):
  File "/usr/lib/python3.10/site-packages/cura/CrashHandler.py", line 396 in _logInfoWidget
  File "/usr/lib/python3.10/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/usr/lib/python3.10/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/usr/bin/cura", line 162 in exceptHook
  File "/usr/lib/python3.10/site-packages/UM/Qt/ListModel.py", line 105 in setItems
  File "/usr/lib/python3.10/site-packages/cura/Machines/Models/ExtrudersModel.py", line 234 in __updateExtruders
  File "/usr/lib/python3.10/site-packages/UM/Qt/QtApplication.py", line 396 in exec_
  File "/usr/lib/python3.10/site-packages/cura/CuraApplication.py", line 877 in run
  File "/usr/bin/cura", line 236 in <module>

fieldOfView, 2022-01-22:
Try the AppImage instead of the package you use now. 
https://github.com/Ultimaker/Cura/releases/tag/4.13.0

fvrmr, 2022-01-24:
Closing this issue since this is not our version of Cura.

closed 2022-01-24 by fvrmr

Arjow, 2022-01-24:
I have the same problem with package installed with pacman but also with the appimage from ultimaker.
with Archlinux (arcolinux)        Kernel: 5.16.2-arch1-1

There is a little difference in the log file python3.10 -> python 3.8 between the appimage and the pacman(arch) version

**Error traceback:**
Traceback (most recent call last):
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/cura/Settings/ExtruderManager.py", line 425, in getExtruderHasQualityForMaterial
KeyError: 'empty_material'

**Logs:**
Thread 0x00007f187d427640 (most recent call first):
  File "/tmp/.mount_Ultima8WgWTM/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f187e429640 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 306 in wait
  File "/srv/cura-build-environment/lib/python3.8/site-packages/zeroconf/__init__.py", line 2534 in wait
  File "/srv/cura-build-environment/lib/python3.8/site-packages/zeroconf/__init__.py", line 1715 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f187ec2a640 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 306 in wait
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 558 in wait
  File "/tmp/.mount_Ultima8WgWTM/usr/bin/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f187f42b640 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/site-packages/zeroconf/__init__.py", line 1369 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f187fc2c640 (most recent call first):
  File "/tmp/.mount_Ultima8WgWTM/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f1885d87640 (most recent call first):
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f1886588640 (most recent call first):
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f18beffd640 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 302 in wait
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 124 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f18bf7fe640 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 302 in wait
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 124 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap
[... 34 lines cut ...]
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/cura/Machines/ContainerTree.py", line 181 in run
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 133 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

[... cut ...]
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/Qt/QtApplication.py", line 396 in exec_
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/cura/CuraApplication.py", line 877 in run
  File "/home/ultimaker/src/output/build/inst/bin/cura", line 236 in <module>
  File "/srv/cura-build-environment/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36 in run
  File "/srv/cura-build-environment/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74 in run

Ghostkeeper, 2022-01-28:
I think what happened is that the Pacman package you used doesn't have any material profiles (missing fdm_materials repository). Cura created a printer then and tried to associate the preferred material with it. However that preferred material doesn't exist. It then falls back to `empty_material` which is not allowed for that printer since it is set to have material (and so it is not in the container tree, which is where the key error occurs).

When you change to the AppImage, even though the materials are present then, the material in the printer is still set to `empty_material` which is still not in the allowed materials for the printer, so it still crashes.

I am unable to reproduce this issue just using the AppImage. Perhaps you should remove the invalid configuration created by the Pacman package and then try only using the AppImage.

Naheel-Azawy, 2022-01-30:
Hi,
I had the same issue on arch using the package from pacman.
This is probably not a solution, but it seems like removing it and installing the appimage does the job for now.
You can check `cura-appimage-bin` from AUR if you like.

Arjow, 2022-02-01:
I removed Cura installed by pacman but the appimage give the same error
Cura-appimage-bin from aur give also the same problem.

Cura old from the uar works but that's a little bit old. 

I think the best thing for me to do is switch to an other slicer. I only need g-code.

NanzeRT, 2022-03-16:
Same issue for me. Fixed it by deleting user data.
Checked on versions from arch repos, AUR and release from git.

NanzeRT, 2022-03-16:
It also fixed when I deleted `~/local/share/cura/4.13/extruders/creality_base_extruder_0+%232.extruder.cfg`