Ultimaker/Cura #11317: Cannot upload to printer

NikAWing, 2022-01-22:
### Application Version

4.13.0

### Platform

Windows 7

### Printer

RRF Duet3

### Reproduction steps

Hey all,

I'm kinda new to reporting bugs or occuring errors, so I hope I'm doing it right.

This problem started for me at least 3 Cura releases ago.
I have a printer with Duet3 controller and RRF. I use the DuetRRF plugin to send g-code files to the printer.
This worked before but suddenty it makes Cura crash (Windows shows Qt5network.dll in the crash information).

so it's Slice -> Upload to printer button -> filename ok -> crash

I can access the printer via webbrowser and also upload files using the web gui fine.

I tried it with disabled firewall etc without any effect.

With best regards,
Nik

### Actual results

Cura crashes

### Expected results

uploading should work without a crash

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

the end of stderr.log I see this:

> 2022-01-22 14:33:06,486 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [764]: Slicing took 1.2980740070343018 seconds
2022-01-22 14:33:06,486 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [765]: Number of models per buildplate: {0: 1}
2022-01-22 14:33:06,487 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [779]: See if there is more to slice...
2022-01-22 14:33:06,490 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
2022-01-22 14:33:06,493 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2022-01-22 14:33:06,497 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2022-01-22 14:33:06,509 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend finished slicing. Resetting process and socket.
2022-01-22 14:33:07,255 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.020000934600830078 seconds
2022-01-22 14:33:10,080 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [170]: Nik Duet 3-upload | Filename set to: Nik Duet3 - ABS_Boden.gcode
2022-01-22 14:33:10,081 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [118]: 'info/send_slice_info' is turned off.
2022-01-22 14:33:10,081 - DEBUG - [MainThread] ArcWelderPlugin.ArcWelderPlugin._filterGcode [186]: ArcWelder is not enabled
2022-01-22 14:33:10,082 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [183]: Nik Duet 3-upload | Loading gcode...
2022-01-22 14:33:10,281 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [194]: Nik Duet 3-upload | Connecting...
2022-01-22 14:33:10,334 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [202]: Nik Duet 3-upload | rr_connect failed with errorCode 203
2022-01-22 14:33:10,335 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [204]: Nik Duet 3-upload | errorCode indicates Duet3+SBC - let's try the DuetSoftwareFramework API instead...
2022-01-22 14:33:10,504 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: Nik Duet 3-upload | Uploading...
Windows fatal exception: access violation

> Thread 0x00002fb8 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.13.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
[... 60 lines cut ...]
> Current thread 0x000028d8 (most recent call first):
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 396 in exec_
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 877 in run
  File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 236 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

cura.log has this at the end:

> 2022-01-22 14:33:06,486 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [764]: Slicing took 1.2980740070343018 seconds
2022-01-22 14:33:06,487 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [765]: Number of models per buildplate: {0: 1}
2022-01-22 14:33:06,487 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [779]: See if there is more to slice...
2022-01-22 14:33:06,490 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
2022-01-22 14:33:06,493 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2022-01-22 14:33:06,497 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2022-01-22 14:33:06,509 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend finished slicing. Resetting process and socket.
2022-01-22 14:33:07,256 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.020000934600830078 seconds
2022-01-22 14:33:10,080 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [170]: Nik Duet 3-upload | Filename set to: Nik Duet3 - ABS_Boden.gcode
2022-01-22 14:33:10,081 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [118]: 'info/send_slice_info' is turned off.
2022-01-22 14:33:10,081 - DEBUG - [MainThread] ArcWelderPlugin.ArcWelderPlugin._filterGcode [186]: ArcWelder is not enabled
2022-01-22 14:33:10,082 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [183]: Nik Duet 3-upload | Loading gcode...
2022-01-22 14:33:10,281 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [194]: Nik Duet 3-upload | Connecting...
2022-01-22 14:33:10,334 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [202]: Nik Duet 3-upload | rr_connect failed with errorCode 203
2022-01-22 14:33:10,335 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._check_duet3_sbc [204]: Nik Duet 3-upload | errorCode indicates Duet3+SBC - let's try the DuetSoftwareFramework API instead...
2022-01-22 14:33:10,504 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: Nik Duet 3-upload | Uploading...

fieldOfView, 2022-01-23:
> I'm kinda new to reporting bugs or occuring errors, so I hope I'm doing it right.

It is a fine report. The only thing is that this issue can only be solved by the creators of the DuetRRFPlugin, so it needs to be reported here instead:
https://github.com/Kriechi/Cura-DuetRRFPlugin/issues

NikAWing, 2022-01-23:
ahh thanks! I was not sure where it goes and the qt5 dll made me think it's rather a problem of Cura itself :)

Kriechi, 2022-01-23:
@fieldOfView the error reported is not really pointing to any problem or code in the plugin:
`Windows fatal exception: access violation`

Have you seen such an error message before that could help me narrow down the search?

ref and further discussion at https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/40

fieldOfView, 2022-01-23:
By that logic, the error is also not in the Cura code. Errors like this are hard to track down, short of running Cura from source in a debugger. Something in PyQt crashes, because of how it is used. For example writing to an object that has already been destroyed can lead to an access violation.

fvrmr, 2022-01-24:
Closing this issue since this is not an issue with Cura.

closed 2022-01-24 by fvrmr