Ultimaker/Cura #11318: Extruder quantity cannot be changed > 1 :  SeeMeCNC Artemis 300 and Robostock Max V3.2

leptoid, 2022-01-22:
### Application Version

4.6 - 4.13.0

### Platform

ALL

### Printer

SeeMeCNC Artemis 300 and Robostock Max V3.2

### Reproduction steps

1.  Add printer.
2. Select Add a non-networked printer
3. Choose either printer model SeeMeCNC Artemis 300 or Robostock Max V3.2
4. Click Machine Settings
5. Attempt to set number of extruders to 2.  Only 1 is allowed via the drop down box.

Reference Code for bad config with missing additional extruder component.
[seemecnc_artemis.def.zip](https://github.com/Ultimaker/Cura/files/7919455/seemecnc_artemis.def.zip)

### Actual results

Printer adds with single extruder.
Attempts to change quantity of extruders fail, as built in configs only allocate space for 1 extruder though devices have 2.

### Expected results

After adding device and clicking on machine settings, number of extruders should be allowed to be set to 2 and each extruder individually configurable.  Currently this is only possible if you do not use the built in profile, but instead make the printer from scratch.

I attempted to modify the resource definition file for the applicable items
```
Windows:  C:\Program Files\Ultimaker Cura 4.13.0\resources\definitions\seemecnc_artemis.def.json
                 C:\Program Files\Ultimaker Cura 4.13.0\resources\definitions\seemecnc_v32.def.json
Source Location:  https://github.com/Ultimaker/Cura/blob/master/resources/definitions/seemecnc_artemis.def.json
                           https://github.com/Ultimaker/Cura/blob/master/resources/definitions/seemecnc_v32.def.json
```
but when I modify it from:
```
        "machine_extruder_trains":
        {
            "0": "seemecnc_artemis_extruder_0"
        }
```

to 
```
        "machine_extruder_trains":
        {
            "0": "seemecnc_artemis_extruder_0",
            "1": "seemecnc_artemis_extruder_1"
        }
```
I thought this change would allow for the second extruder, but instead it is apparently incomplete because clicking add will then result in the printer failing to add at all (have to save and restart software for change to take effect... then you find out it doesn't work).  Clicking the add button fails to do anything.  

I do not understand the syntax/dependency linking enough to know why.
[cura.log](https://github.com/Ultimaker/Cura/files/7919473/cura.log)
 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

SeeMeCNC Artemis 0.5 Nozzle in log is manually added config.  Works fine.

I think I forgot a comma in the modded code when this happened... or it is when I click "add" and it does nothing (after modding for 2 extruders)
`2022-01-16 11:55:08,162 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [300]: The definition cache for definition seemecnc_artemis failed to pickle.`

[image]

leptoid, 2022-01-22:
p.s.  There is no project file attached because that would imply a person would get to that point in time.  I could definitely print with a single extruder... but then why do I have 2 if 1 can't be used? 

[image]

leptoid, 2022-01-22:
Were the functioning 3mf profiles which would add and configure the printers correctly, though the default deck would not be as "pretty":  
[Artemis 300 and  RostockMAX Profiles (Working).zip](https://github.com/Ultimaker/Cura/files/7919488/Artemis.300.and.RostockMAX.Profiles.Working.zip)

GregValiant, 2022-01-23:
You have to add a second extruder file.
Extruders are in the "resources\extruders" folder.  You can make a copy of your "seemecnc_artemis_extruder_0.def.json" file and name it "seemecnc_artemis_extruder_1.def.json".  You will then need to open it and change a couple of lines in it for the extruder name , extruder position, and extruder number.
"name": "Extruder 2"
"position": "1"
"extruder_nr": { "default_value": 1 }
There might be secondary things that are involved in adding an extruder.  Nozzles, quality settings...things like that.

nallath, 2022-01-24:
Did you already check the various wiki pages about the structure for machine files?

See: 
https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura
https://github.com/Ultimaker/Cura/wiki/Definition-Files-Explained

leptoid, 2022-01-24:
I did not.  I just foubd it interesting that even in version 3 this was not correctly configured for default adding.

I will read over that stuff and either make a proper config for the printer that actually matches its capabilities, or give up abd just keep doing it how i make it now.

Thanks for each of your advice.  Guess we will see if I get the time to fix the bad plugin.

GregValiant, 2022-01-24:
When you added the line "1": "seemecnc_artemis_extruder_1" to the definition file it caused Cura to go looking for the extruder file "**seemecnc_artemis_extruder_1.def.json**" and Cura couldn't find it.  That generated the error.

When you add the definition for the second extruder it should become available in Cura.

64bittuning, 2022-01-28:
@leptoid 
Under /resources/extruders/
create a new file called "seemecnc_artemis_extruder_1"

Add the following inside the file:

{
    "version": 2,
    "name": "Extruder 1",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "seemecnc_artemis",
        "position": "0"
    },

    "overrides": {
        "extruder_nr": {
            "default_value": 1,
            "maximum_value": "1"
        },
        "machine_nozzle_size": { "default_value": 0.5 },
        "material_diameter": { "default_value": 1.75 }

    }
}

Now open /resources/extruder/seemecnc_artemis_extruder_0

Scroll down to the "overrides" segment and change it with:

    "overrides": {
        "extruder_nr": {
            "default_value": 0,
            "maximum_value": "1"
        },
        "machine_nozzle_size": { "default_value": 0.5 },
        "material_diameter": { "default_value": 1.75 }

    }
}

Now open /resources/definitions/seemecnc_artemis.def.json and replace the "machine_extruder_trains" section with:

        "machine_extruder_trains":
        {
            "0": "seemecnc_artemis_extruder_0",
			"1": "seemecnc_artemis_extruder_1"
        }

restart Cura and enjoy the use of your second extruder!

[image]