Ultimaker/Cura #11319: Tree Support not finding path to overhang

struandw, 2022-01-23:
### Application Version

4.12.1

### Platform

macOS Big Sur, M1 2020

### Printer

Ender 3 Pro

### Reproduction steps

1) Import model (stl is attached) and position on build plate.
2) Enable supports and select *Tree*. Set *Support Placement* to *Touching Buildplate*.
3) Slice.

### Actual results

Overhangs which clearly need support and are supported when *Support Placement* is set to *Everywhere* are not supported. No support material is generated.

### Expected results

Support material should be generated, similarly to that generated when *Support Placement* is set to *Everywhere*.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[CE3PRO_VFP Resistance is Character-Forming-18.3mf.zip](https://github.com/Ultimaker/Cura/files/7920459/CE3PRO_VFP.Resistance.is.Character-Forming-18.3mf.zip)
[image]
[image]
[VFP Resistance is Character-Forming-18.stl.zip](https://github.com/Ultimaker/Cura/files/7920460/VFP.Resistance.is.Character-Forming-18.stl.zip)

fvrmr, 2022-01-27:
Hi @struandw thank you for your report. 
This is how it should work on your model. Touching buildplate means that it will support overhangs above the buildplate and with your model right now you don't have an overhang above the buildplate. So that's why it will work with everywhere because then it will support everywhere where the model has an overhang. 
Read here more about the support placement: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_type.md

I will close this issue, since this is not a bug. But reach out if you have questions.

Ghostkeeper, 2022-02-01:
You'd expect that at least some of those branches could reach the build plate though.

"Touching build plate" means that all support must rest on the build plate. But if the support can reach around and support more (with either Conical Support or Tree Support), it should.

This is a simple test case I've been using when tree support was created which shows the behaviour:
[image]

I can't figure out why this doesn't work for your model. I've seen before that a branch can get accidentally pulled in over the border into the collision zone of a mesh (from where it's impossible to avoid hitting the model according to the branch angle) but not this much.
Ironically, reducing the Tree Support Branch Angle (to like 20) improves it, while in theory you'd expect less of the model to be reachable from the build plate then.

struandw, 2022-02-01:
Thank you for clarifying this; I thought I was misunderstanding something about the Branch Angle.

If you need any more information, please let me know! Weird bugs like this are my jam. 

Best,
Struan Duncan-Wilson

DerGenaue, 2023-07-04:
This issue should be solved in the new Cura 5.4 release