Ultimaker/Cura #11320: Cura 4.13.0 Thin wall issue

DVSVIDEO, 2022-01-23:
### Application Version

4.13.0

### Platform

Windows 10 64bit

### Printer

Micromake C1 modified core xz printer

### Reproduction steps

Download this: https://www.thingiverse.com/thing:3845689
Load stl file: silica_Gel_Body on Cura
settings: PLA 0.4 Draft 0.2mm, 4 top, 4 bottom layer, 2 walls
press slice button

### Actual results

It uses a bottom or top infill instead of a thin wall.
The cura misrecognises the thin wall and uses bottom/top fill, which gives a strange result.
[image]

If I turn off the thin wall option, this part disappears from the model.
[image]

Here it is up close, without the layers above:
[image]

### Expected results

Correctly interpreted thin wall in SuperSlicer:
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Silica_Gel_Body.zip](https://github.com/Ultimaker/Cura/files/7920707/Silica_Gel_Body.zip)

fieldOfView, 2022-01-23:
Try the Cura Arachne beta, which will do a better job at slicing thin walls:
https://ultimaker.com/learn/3d-printing-revolution-continues-second-beta-for-ultimaker-cura-arachne
This is a preview of the slicing improvements that will be part of Cura 5.0

DVSVIDEO, 2022-01-24:
> Try the Cura Arachne beta, which will do a better job at slicing thin walls:

I tried it, but I didn't like it.
The outer walls of the 20x20 test cube were not pretty. 4.13.0 gave a much nicer print.

GregValiant, 2022-01-26:
Can you post a 3mf file of your project (File | Save Project)?  I can't duplicate this in 4.12 or 4.13.
This is with my plain old regular settings.  Print thin walls is turned off.
[image]

fvrmr, 2022-01-31:
Hi @DVSVIDEO thank you for your report. Like @GregValiant I could also not reproduce your issue with the stl and reproduction steps.
Also @fieldOfView is right that thin walls are sliced better in Arachne since it uses variable line width.

Ghostkeeper, 2022-02-01:
The entire feature has been removed in favour of variable line width in Ultimaker/CuraEngine#1210 . We consider this fixed for the next release. If you are having trouble with the Arachne beta, we'd love to hear about those instead. That change is not going to get reverted.

This is effectively a duplicate of https://github.com/Ultimaker/Cura/issues/5110.

closed 2022-02-01 by Ghostkeeper