Ultimaker/Cura #11322: Automated Lay Flat and Select Face

Firedogs2x, 2022-01-23:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Modix Big-60

### Reproduction steps

1. I opened a stl file that was a 20mm square x .2mm thick part.
2. The model came in on its side (.2mm edge).
3. I selected the part and then I selected "Lay Flat".
4. The part rotated.
5. I selected "Slice".
6. I saved the gcode and ran the file on my printer.
7. On my printer it printed the skirt and then started printing the part. The first section of the print was 20mm x 10mm. Then it attempted to print the second half. The second half was next to the first and over the skirt.
8. When I looked at the preview it shows the part as if it is two sections. One section inside of the skirt and one section crossing over the top of the skirt.
9. I have seen this issue on software called Blade. It is CURA but modified to work on BigRep printers. With that software I had the same issue but there I was using "Select face to align to the build plate".

[image]
[image]
[20 x 20 x .2mm THK Lay Flat.3mf.zip](https://github.com/Ultimaker/Cura/files/7921273/20.x.20.x.2mm.THK.Lay.Flat.3mf.zip)
.

### Actual results

They have been stated in the reproduction steps.

### Expected results

The expected results should have been a 20mm square x .2mm thick part with a skirt printed outside of the completed part.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Everything should be in the Reproduction Steps.

The only way I have found to solve this is to delete the current part in CURA.
Then open the part in CURA.
Then manually rotate the part 90 degrees.
Then run "Slice".
And now I have a good file to print with.

Thank you for your software and your help.

fieldOfView, 2022-01-23:
The Lay Flat function is fairly naive. I can say that, because I am the one that ported it over from Cura 15.4 five and a half years ago (as my first functional change to Cura) and AFAIK it has not changed since. It tries to find 3 points that are "some distance apart" and make up a plane without another vertex "protruding" from it. 

If the model is not 100% flat, then it may end up orienting it on a plane such that only part of the "not 100% flat" plane is actually touching the buildplate. A rectangle where one vertex is actually 0.01 mm off-plane from the other 3 will cause half the plane to be off. This is likely what happened here.

Firedogs2x, 2022-01-24:
I made the model myself in SolidWorks. So unless the stl was not made totally flat that was not the issue.

 

What about the select a face is that the same?

 

Thank you for the prompt reply.

fieldOfView, 2022-01-24:
Looking at the model in a text-editor, I notice that the model has no thickness/height; all of its z-coordinates are the same. For 3d printing things need to have a thickness. To be fair, I don't understand how this would confuse the lay flat function to the point of putting the model at an angle.

Firedogs2x, 2022-01-24:
The model has a thickness of .2mm as stated in the initial comment.
And as stated when the model first is brought into CURA it is positioned on the .2mm edge.
When I created the part all I did is draw a sketch on the Top plane and extrude it .2 mm.

Ghostkeeper, 2022-02-01:
The model seems correct to me. It's completely flat on its two big sides, and it has a height of 0.2mm according to Blender as well as Cura.

When I use the "Lay Flat" feature this issue appears. When I Lay Flat On Face for one of the two big faces, it works correctly.

It's like Lay Flat selected one of the higher vertices to lay the object flat on.

fvrmr, 2022-02-04:
We have discussed this with the team and decided to defer this issue, since it is an edge case and it will not get a high priority to fix for us. I hope you understand.

Firedogs2x, 2022-02-04:
I understand.

But please keep it in mind to consider trying to make it more reliable in the future.

Thank you.

closed 2023-06-10 by github-actions[bot]