Ultimaker/Cura #11329: (4.13.0) SentryLogger crashes on KDE Plasma 5.23.5

thecow275, 2022-01-24:
### Application Version

4.13.0

### Platform

Linux 5.16.2-1-MANJARO (64-bit)

### Printer

Creality Ender 3

### Reproduction steps

install cura and dependencies from the community repo and launch cura.

### Actual results

Sentry Logger fails to load under kernel 5.16.2 on kde plasma 5.23.5 (KDE framework 2.90.0).

### Expected results

Sentry Logger loads correctly

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7927193/cura.log)

MichielioZ, 2022-01-27:
I got the same plugin failure on my Anycubic Mega Zero 2.0 after Pamac upgraded Cura to 4.13.0
I'm on Manjaro too with 5.15.16-1 kernel, also using the community repo.
The UFPWriter and UFPReader plugins also crashed, but after a search on internet found that I should install libCharon which fixed that problem. (would have been nice not to have the problem in the first place though)

nallath, 2022-01-27:
We don't manage or support the community repo. Please contact them if there is an issue with the build that is provided there.

The releases we create are managed via Appimage which includes all the dependencies required to run.

closed 2022-01-27 by nallath