Ultimaker/Cura #11332: Sprial Lines are in Segments

zwu513, 2022-01-25:
### Application Version

1.12.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Slice File
2. Preview
3. Print

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CE3_TestSweepPattern_0.28mm_15%_215C_60C_0H_2M.zip](https://github.com/Ultimaker/Cura/files/7930654/CE3_TestSweepPattern_0.28mm_15._215C_60C_0H_2M.zip)

fvrmr, 2022-01-25:
Hi @zwu513 thank you for your report.
If you slice your model in our Arachne engine it will slices, like you expected. This is because Arachne has a variable line width. And your model is probably to thin to print, that's why you will get these gaps. 
https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta_2

[image]

zwu513, 2022-01-25:
The confusing part is that I've gotten it to slice as a single line with no issues. It's seems it had something to do with the exact width and height. I'll try the beta, thank you.

Ghostkeeper, 2022-02-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/4685. This is fixed by #1210 .

closed 2022-02-01 by Ghostkeeper