Ultimaker/Cura #11333: Cura Slicer fails to print model Spiralized when requested

mvandere, 2022-01-25:
### Application Version

4.13.0 and 4.12.1

### Platform

Windows 10

### Printer

Crealty Ender V2 + BCN3D

### Reproduction steps

Created a shape like an open ended light globe and set to print 'Spiralize outer contour'.
[GlobeSample.zip](https://github.com/Ultimaker/Cura/files/7931197/GlobeSample.zip)

### Actual results

Model prints a layer at a time

### Expected results

Model should be spiralized / print in Vase mode

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This project prints properly using simplify3D Slicer.

fvrmr, 2022-01-25:
Hi @mvandere thank you for your report.
Could you share a project file (file - save project)?

mvandere, 2022-01-25:
[GlobeSamplestl.zip](https://github.com/Ultimaker/Cura/files/7937200/GlobeSamplestl.zip)
Sorry about that, was using Fusion and 'transparently' passing to Cura, so didn't know the required file format

mattwn, 2022-01-26:
Hi @mvandere, you need to do the file->save project in cura. The resulting file type will be a .3mf. It will contain the model itself and your slicer settings. Both of the files you attached are just models, the slicer settings will also be needed.

I was able to slice your model with Spiralize enabled, so it's potentially something in your slicer settings that's the issue.

GregValiant, 2022-01-26:
That STL is hollow like a lightbulb.  It won't print in spiralize.  Because it's a single wall it looks like spiralized worked but I get a Z seam.

mvandere, 2022-01-27:
The initial problem was that 'bottom layers'/'initial bottom layers' were set to large numbers. It appears that Cura will not spiralize for the number of layers set there.
Surely a message to that effect should be produced and/or the layers set to zero when spiralize is used.
Now there's a problem with the quality of the slicing produced, there are 'gaps'. If you walk through the created file you will see it seems to spiralize until about 1/3 of the way up, then it starts dancing a jig, changing direction all over the place until it settles down again. There's a screenshot of Cura in this area in the zi
[Globe.zip](https://github.com/Ultimaker/Cura/files/7948894/Globe.zip)
p

mattwn, 2022-01-27:
You can ALMOST Spiralize a thin wall using Surface Mode: Surface with Spiralize. When you try this, the G Code has the nice smooth incrementing Z moves with each G1. However, it still decides to put in a layer change travel and retraction (if enabled). 

Also, Surface Mode: Surface slices physically larger than the actual print. It looks like it's oversized by about 1 wall thickness, or the difference between an Inclusive/Exclusive tolerance, however changing the tolerance setting doesn't fix the oversize surface.

mvandere, 2022-01-29:
"You can't spiralize a thin wall model. It doesn't work because the inside wall has air on one side, and the outside wall has air on one side. That means there are two walls." On my model the thin wall is the width of the 'pen', so there should only be a single wall that can be spiralised. It shows this clearly in Cura. The problem is that part way up it starts changing directions and missing 'bits' (See the pic in globe.zip from 2 days ago), for the most it spiralizes.

"You can see the red dot on the top indicating it needs support." A. if you look at the model you can clearly see it does not need support, maybe you have some different setting that means support is required. B. in my Cura there is no red dot, as I'd expect.

"But you could do it as a Surface" If I tell it to do it as a surface it does two layers, and hence it 'inherently' disables spiral mode.

Maybe Cura just 'assumes' that you need a base to spiralize, and gets it wrong here because there is no base?

Using the Simplify 3D slicer it produces similar errors to Cura in the middle of the print. However, if I set on the "Merge all outlines into a single solid mode" option it prints a perfect spiralized Globe.

Be really nice if I could get it to work in Cura.

mvandere, 2022-01-29:
Actually, try playing with the 'Slicing Tolerance' setting. Changing it to 'exclusive' results in a 'useless' print. Setting it to 'Inclusive' produces a slightly better print, but at the same time the errors are more obvious, if viewed from the inside.

Ghostkeeper, 2022-02-01:
The issue really has nothing to do with overhang or anything. Cura makes no assumptions about overhang in spiralise mode. It'll just slice it, and if you think the overhang is fine, it'll let you print that. It also has nothing to do really with needing a base.

The issue is simply that spiralise mode does assume that it can print the entire model in one spiral. With hollow objects, that is just untrue, as it will attempt to print both walls. The inside wall as well as the outside wall.

To discard the wall on the inside, you could enable "Remove All Holes". This removes all internal cavities, and technically makes the inside of this bulb filled in. Without spiralise mode this would make it fill that with infill, but spiralise mode of course makes the model hollow.

I think this is a matter of misunderstanding what spiralise mode does. That is something we could try to make more clear, but there's limited space in the descriptions. The Settings Guide explains it a bit more extensively.

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/9190.

closed 2022-02-01 by Ghostkeeper