Ultimaker/Cura #11337: Support Blocker not working!!!

Dudelsack3HD, 2022-01-25:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

I added a cylinder-shaped model and set its mesh type to : Don't support overlaps.

### Actual results

When slicing the model and inspecting it, it appears like the support blocker has no effect whatsoever.

### Expected results

The support structure within the support blocker model should not exist

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[AV_Body1.zip](https://github.com/Ultimaker/Cura/files/7934833/AV_Body1.zip)

fieldOfView, 2022-01-25:
"Don't support overlaps" meshes only "mark" things in your normal object not to get support generated for it. They are not objects that are somehow "cut" from all supports. They have to overlap the area of the model that you don't want to get supports (ie: the tops of the support structures).

"Do not create support for this overhanging part", instead of "remove all support here".

GregValiant, 2022-01-25:
_"They have to overlap the area of the model that you don't want to get supports"_
Whoop - there it is.  Your blocker isn't situated to do anything.
The bottom of that inside roof (that wants support) is at Z=84.  Your Initial Layer Height is 0.3 and so Cura can't hit 84.0 and the layer goes down at 84.1.  Your support blocker is only 83mm tall and is sitting on the build plate.  Consequently it isn't blocking anything because it's not touching the "red" area on the underside of that roof .

Make that Support Blocker 0.2mm tall and locate the bottom at Z=84 and see what the slice looks like.  Then move it down to Z=83.9 and slice again and check the preview.

Dudelsack3HD, 2022-01-26:
Thank you very much! This solved the problem.

closed 2022-01-26 by Dudelsack3HD