Ultimaker/Cura #11338: Solid Bottom/Top in Surface Mode

Repoleved247, 2022-01-25:
### Is your feature request related to a problem?

I need internal structures for lightweight models such as planes, drones etc. that are defined by myself. This can be achieved in surface mode, but I also need solid bottom and top surfaces. This would help tremendously with stability and would make it easier to glue parts together. None of the existing modes are suitable for this problem.

### Describe the solution you'd like

Solid Top/Bottom Layer Inputs shouldn't be ignored by the surface mode.
Bottom and Top Layers act as in normal mode, everything in between like surface mode.

### Describe alternatives you've considered

Give us the option to change profiles at a specific Z-height.
This is a more complex solution, but I think more people would benefit from this kind of feature.

### Affected users and/or printers

Changing profiles at a specific Z-height would affect anyone, who want to use less material for large prints because you could change the infill for sections with more or less detail.
You could also improve speed and optimize profiles for specific sections of your print.

### Additional information & file uploads

[image]

mattwn, 2022-01-26:
I second this. While you can change a few parameters at Z height with the post processing script, and you can change some parameters with cutting meshes, it's all pretty limited. Cutting meshes only allows you change a subset of parameters and the rest will follow the global profile no matter what the cutting mesh asks for.

Being able to define (for example):

- 0 to 20 mm of Z: use profile I
- 20 to 50 mm of Z: use profile J
- 50 to end mm of Z: use profile K

Where all parameters are respected and nothing (or as little as possible) defaults to the global, would be very useful.

Failing that, letting the cutting meshes change all (or at least a larger subset) of the parameters would work as well.

fvrmr, 2022-01-31:
Hi @Repoleved247 thank you for your feature request.
Like @mattwn mentioned this is indeed a duplicate. See https://github.com/Ultimaker/Cura/issues/4235
I will close this issue, please continue the conversation in the other thread. So we keep everything in one post.

closed 2022-01-31 by fvrmr