Ultimaker/Cura #11340: [FR] Import/export position of Objects and Groups

tispokes, 2022-01-26:
### Is your feature request related to a problem?

When I print e.g. 40 small equal items. I arrange them as I like to. Fine okay. But when I now change something on my original step-file and create a new stl, I have to do these steps again. The 3mf file does not help me in this case.

### Describe the solution you'd like

Would be nice to export and import a csv like file with the arranged items on the plane. Also faster, because you can compute the exact position.

### Describe alternatives you've considered

none found

### Affected users and/or printers

Users who print a specific part a lot on one build plate would may like this improvement.

fvrmr, 2022-01-31:
Hi @tispokes thank you for your report. 
If you update the file that is loaded in Cura you can reload your models (F5). So for example I have 10 cubes named cube.stl, I make a change to cube.stl and save it under the same name. You can right click and select reload your models or press F5. Now all my 10 cubes are updated with my recent changes.
Let me know if that would work for you.

tispokes, 2022-01-31:
H it's reloading from the source. Ah sure, where else. Okay maybe from RAM. But not really obvious. Thanks!

tispokes, 2022-03-10:
@fvrmr I just got over this again. So I wanted to reload my models. But I have no clue which version of the file (the specific filename) I have inside the 3mf. Would be nice to see somehow which STLs (including full path) are inside a 3mf. Hope you understand what I mean.

tispokes, 2022-03-10:
okay with lib3mf it's possible. integrated in cura would be nice

Ghostkeeper, 2022-03-20:
That is a duplicate of https://github.com/Ultimaker/Cura/issues/7680 then.

closed 2022-03-20 by Ghostkeeper