Ultimaker/Cura #11341: Cura mouse buttons configuration

DVSVIDEO, 2022-01-26:
### Is your feature request related to a problem?

The middle button on my mouse (the scroll button)
has been broken for some time now, so moving the view is impossible for me.

### Describe the solution you'd like

I would like to ask the developers to make the model view movement in Cura the same as in SuperSlicer/PrusaSlicer and Simplify3D slicers:
-left mouse button = rotate
-right mouse button = move up -down
-mouse scroll = zoom in
It would also be nice if all this could be configurable

### Describe alternatives you've considered

Since many people are accustomed to the current button assignment/controls of the Cura, it could be possible to choose between the original and my proposed

### Affected users and/or printers

It would be useful for everyone who uses more slicers, because it would be more uniform in use, it would speed up the work

fvrmr, 2022-01-31:
Hi @DVSVIDEO thank you for your feature request.
This is a duplicate of https://github.com/ultimaker/cura/issues/9062 and this is related to a ticket on our backlog.
For now you can also hold the space bar and move your mouse to scroll. So you don't need your scroll button to zoom.

I will close this issue since it is a duplicate.

closed 2022-01-31 by fvrmr

Ghostkeeper, 2022-02-02:
Note that you can hold the space bar to rotate and zoom as well. This was originally intended for people to work without a scroll wheel or on a touch pad.