Ultimaker/Cura #11343: Add {material_weight} replacement pattern

pedrolamas, 2022-01-26:
### Is your feature request related to a problem?

Cura already calculates and shows the material weight, but as far as I can see, this is not output on the gcode files and there is no pattern available for it.

[image]

This is useful information for Klipper users as it will allow to show this information in the web interface (via Moonraker)

### Describe the solution you'd like

For a new `{material_weight}` replacement pattern to be available.

### Describe alternatives you've considered

None available AFAIK...

### Affected users and/or printers

All?

fieldOfView, 2022-01-26:
Unfortunately this can't work in Cura. The replacement patterns are processed by Cura before slicing, and the material weight is only available _after_ slicing.

pedrolamas, 2022-01-26:
Oh, that's a pitty... 😕

tispokes, 2022-01-27:
@pedrolamas 
I have this in my start G code and it works fine.
`;Filament weight = {filament_weight}`

pedrolamas, 2022-01-27:
Oh, that DOES work!! 😁

[image]

And it matches the Cura interface, so it seems all good!

@fieldOfView guess this is already available?

tispokes, 2022-01-27:
> Unfortunately this can't work in Cura. The replacement patterns are processed by Cura before slicing, and the material weight is only available _after_ slicing.

Why it works then?

fieldOfView, 2022-01-27:
I don't know. There must be something (a plugin) that runs after Cura is done slicing.

pedrolamas, 2022-01-27:
I just confirmed on a brand new install (via Windows Sandbox) with no extra plugins added, seems to work fine even there!

[image]

No idea how or why this works, but seems to extra plugin is required! 😁

pedrolamas, 2022-01-27:
Closing the issue as this is already available right now via `{filament_weight}`

Thanks for the help @tispokes and @fieldOfView!

closed 2022-01-27 by pedrolamas

fieldOfView, 2022-01-27:
I have added the additional replacement patterns I found to http://files.fieldofview.com/cura/Replacement_Patterns.html

pedrolamas, 2022-01-27:
Awesome stuff @fieldOfView , that is exactly where I go when I want to see what replacement patterns Cura has to offer! 😁