Ultimaker/Cura #11349: Tree support does not correctly apply Z distance and overlaps model

julienlamalice, 2022-01-27:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Generate Support: Yes
2. Support structure: Tree
3. Support placement: Everywhere

### Actual results

A) Tree support does not correctly apply Support Z Distance. The first layer of the support overlaps the top layer of the model, effectively merging with it. Enabling support interface results in the same issue with the first layer of the interface merging with the model.

B) The generated tree support intersects the models at several points, thus not respecting the Minimum Support X/Y Distance.

### Expected results

A) The generated tree support's first layer should be above the top layer of the model, respecting the Support Z Distance setting

B) The tree support branches should not be intersecting the model, they should be avoiding it or not be generated at all.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Giant Spider.zip](https://github.com/Ultimaker/Cura/files/7947077/CE3_Giant.Spider.zip)
[image]
[image]
[image]
[image]

fvrmr, 2022-01-27:
Hi @bagelsesame thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/11258. That's why I will close this issue.

closed 2022-01-27 by fvrmr

Ghostkeeper, 2022-02-03:
This looks more like https://github.com/Ultimaker/Cura/issues/11136, which isn't fixed yet in 4.13.1 (see CURA-8874).