Ultimaker/Cura #11353: Extensions modify scaling properties

cstoribal, 2022-01-27:
### Application Version

4.13.0

### Platform

Linux Xubuntu 18.04 and Windows 10

### Printer

Ender 3

### Reproduction steps

When having installed both extensions: 
Tab Anti Warping and Custom Supports

- Import an object .stl in Cura at scale 100%
- Add any Anti warping Tab or any Custom support that would increase the size of the bounding box of the object

### Actual results

The new scale value of the .stl is modified

### Expected results

Scale value should be linked to the relative scaling of the points in the .stl rather than being recomputed from the size of the bounding box of the object with additional components

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Seems to happen since 13.0, but I'm not sure.

GregValiant, 2022-01-27:
I never noticed that before.  The support blockers don't do it but both the Cylindrical Supports and Anti-Warping tabs do.
(I've always found the cylindrical supports to be a little a little flaky.)

fvrmr, 2022-02-01:
Hi @cstoribal thank you for your report. 
I could reproduce this and also when changing the support blockers to print as support.
I will bring it up with the team. Keep you posted.

fvrmr, 2022-02-11:
I have discussed it with the team and we concluded that this is an issue with the plugins. Because when using a support blocker and select this it print as support, it won't happen. I think I check the support blocker when having already a custom support from the plugin to the model.
So in that case we will not fix this, since we don't maintain these plugins. Please report this to the maintainer of the plugins.

closed 2022-02-11 by fvrmr