Ultimaker/Cura #11354: Infill visible on top surface / outer wall with no support under it

unromeo, 2022-01-27:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

I hope I can explain this so you can understand. The part has a step-over (outer wall, then a small horizontal step, then again outer wall). The second part of the outer wall that is resting on that step, is being printed directly on the infill without any support under it. There are a few supported areas, but not the whole perimeter. 

[image]

This is how it looks on that middle layer : 

[image]
 
As you can see, there are only small areas supporting the wall on the right side. 
This is for a 2 line wall , everything is well if switching to 1 line : 

[image]
[image]

### Actual results

Outerwall sits on visible infill

### Expected results

The slicing for a 1 line outer wall is correct. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_Lego_Track_Curved_w_support_mtu_v2.zip](https://github.com/Ultimaker/Cura/files/7951210/CFFFP_Lego_Track_Curved_w_support_mtu_v2.zip)

GregValiant, 2022-01-27:
Change the "Top/Bottom" - "Skin Removal Width" to 0.4.

unromeo, 2022-01-28:
Well, this helps ! Sorry for wasting your time.. the guys at the Cura Forum thought this was still a bug. But I still do think that the calculated value of 0.8 is wrong and it should default to 0.4. 

Thanks !

closed 2022-01-31 by unromeo