Ultimaker/Cura #11355: File > New Project on a Mac should not "reuse" the current window. It should create a new window hand should be able to have several windows (projects) opened at once

altimac, 2022-01-27:
### Is your feature request related to a problem?

Yes. I often have to compare projects of slices in Cura, typically side by side.
Mac OS is great for that has the usual behavior of Document Based Applications is that you can have several documents (one per window) opened at the same time (think MS Word for example). Then you're free to rearrange the window size to see both windows on your display.
In Cura, I would like to do the same : have 2 projects opened in 2 windows, typically side by side, and interact with both, completely separated.

### Describe the solution you'd like

see above

### Describe alternatives you've considered

none ?

### Affected users and/or printers

Mac users of course, maybe windows users too?

BagelOrb, 2022-01-30:
You can just start Cura up twice right? Or is that not possible on Mac?

fvrmr, 2022-01-31:
Hi @altimac thank you for your feature request.
Like @BagelOrb mentioned you can open Cura twice. Would that solve your request?

altimac, 2022-01-31:
The answer is a bit complex: on a Mac you can't open the same binary/executable or application twice.

fieldOfView, 2022-01-31:
As a workaround, I think you can make a copy of the whole application, and run that as a second one.

It is not really recommended to run two instances of the same version of Cura at once on any platform though. Because they are both writing to the same set of settings files, you can end up with a mix of two sets of settings.

fvrmr, 2022-02-01:
My bad I thought this was possible, but I got mixed up with having different versions of Cura open.

Ghostkeeper, 2022-02-03:
You can indeed have multiple instances of Cura open. If they are the same minor version, they will share a configuration directory. We ensure that file writes are done atomically. However deleting profiles or printers is still "unsafe" then; the other instance will happily put it back in the next autosave if it is touched.

However MacOS indeed prevents the same application to run twice, so I don't think we can implement starting a new Cura instance for new project files. So the only way to implement this would be to have multiple windows/tabs/etc within the same Cura instance. This is https://github.com/Ultimaker/Cura/issues/7009 , but it was rejected due to the high amount of work for relatively small gain.

closed 2022-02-03 by Ghostkeeper

altimac, 2022-02-04:
OK if you say it's safe to have 2 instances of Cura (very same version) opened at the same time, I'll do that. Thanks.

fieldOfView, 2022-02-04:
Just don't expect changes made to settings in one instance to carry over in the other instance, and don't be disappointed if one instance overwrites the settings of the other instance when closed. These issues are the same on all platforms.