Ultimaker/Cura #11358: Not remembering settings

docean24, 2022-01-28:
### Application Version

Arachne_beta_2

### Platform

Windows 10

### Printer

custom

### Reproduction steps

Change material temperature, slice a file, close Cura. Open Cura again and the old temperature is used. Usually the last settings used for slicing are remembered when I open Cura again.[](url)

### Actual results

Old temperatures are used instead of new temperatures.

### Expected results

Usually the last settings used for slicing are remembered when I open Cura again.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2022-01-31:
Hi @docean24 thank you for your report.
I can't reproduce your issue, does it only happen in Arachne?
Did you maybe change profiles?

docean24, 2022-02-02:
Only happens in Arachne, I did not change profiles. I have Cura 12.1, 13.0, and Arachne installed with plugins arc welder 3.5.2, calibration shapes 1.7.0, custom supports 0.1.0, linear advance settings 3.5.11, measure tool 1.0.2, octoPrint connection 3.6.0, startup optimiser 3.5.2, version upgrade 4.9 to 5.0, and z offset setting 3.5.7. Let me know if you need anything else.

fvrmr, 2022-02-11:
Could you maybe share a screencast where I can see that it is not saving your settings?

Ghostkeeper, 2022-02-14:
A log file might also clarify what's happening.

closed 2022-02-25 by no-response[bot]