Ultimaker/Cura #11363: Ending gcode : Z axis moves wrong direction

QuestionDevelopment, 2022-01-28:
### Application Version

4.13.0

### Platform

WIndows 10

### Printer

Ender 5 plus

### Reproduction steps

Print something with following ending gcode

```
G91 ;Relative positioning
G1 Z80 ;Raise Z more
```

### Actual results

Bed gets raised

### Expected results

Bed gets lowered

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Hello everyone,

I have a ender 5 (SKR E3 Mini V2) that works perfect (for a long time) and runs the following end gcode.

G91 ;Relative positioning
G1 E-5 F600 ;Retract a bit
G1 E-2 Z0.2 F600 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z80 ;Raise Z more
G90 ;Absolute positionning
G1 X0 Y0 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z
M77 ;End print timer

However my new ender5+ (SKR V2) is working perfectly EXECPT after a perfect print it runs the z bed up at the end...on a small print it will actually run it at the end of my lead screw. The only other thing I noticed was when I tried to lower the bed manually with the screen it works almost all the time except (1 out of 20 times/randomly) it goes up. I confirmed that I was in relative position (G91) when this happened.

The relevant marlin config settings are the same between both machines

#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false
#define INVERT_E0_DIR true

GregValiant, 2022-01-29:
It would seem that this is between your firmware and your SKR board.  That end gcode should work as you think it should.  It's the same as the definition file except for the "80" where the definition file is "10".
Since you've hard coded in the 80mm then for tall prints there might be a problem as there would be times when it can't move 80 because it's too close to the max height.  It may be that the "machine_height" is defined incorrectly in the firmware.  It should be arresting the movement before the bed over-travels.
The Cura team will take a look, but I don't see any way this is a Cura problem.  You have a 3rd party printer with a 4th party mainboard, that was installed by a 5th party with firmware from a 6th party...and the End Gcode is correct.

QuestionDevelopment, 2022-01-29:
Thanks for your professional handling of this ticket.  You were 100% correct...it ended up being a few overlapping issues but it did not involve Cura.

closed 2022-01-29 by QuestionDevelopment