Ultimaker/Cura #11364: Fehler Beim Einzug

Berndzlin, 2022-01-28:
### Application Version

4.13.0

### Platform

windows11

### Printer

Tronxy xy2 pro

### Reproduction steps

1. Profil mit Einzug = 5mm erstellen
2. G-code erzeugen

### Actual results

vor dem ersten Layer wird das Filament  im G-Code um 5mm eingezogen,
aber nicht wieder vor dem Start des 1. Layers zurückgeschoben.

### Expected results

;FLAVOR:Marlin
;TIME:8257
;Filament used: 3.81035m
;Layer height: 0.2
;MINX:95.5
;MINY:88.377
;MINZ:0.3
;MAXX:159.499
;MAXY:152.081
;MAXZ:14.5
;Generated with Cura_SteamEngine 4.13.0
M82 ;absolute extrusion mode
; XY-2 Start Code
G21 ; Set Units to mm
G90 ; Set Positioning to Absolute
M82 ; Set Extruder to Absolute
M107 T0 ; All Fans Off
M140 S55 ; Heat bed to layer 0 setting
M104 S210 T0 ; Heat nozzle to layer 0 setting
M190 S55 ; Wait for bed to heat
M109 S210 T0 ; Wait for nozzle to heat
G28 ; Home all axes
;G1 Z5.0 ; Raise nozzle to prevent scratching of heat bed
;G1 X0 Y45 ; Move nozzle to Home before heating
;G92 E0 ; Set Extruder position to zero
;G1 X2 Y45 Z0.2 F1500.0 ; Move to purge line start position
;G1 X2 Y160 Z0.2 F1500.0 E20 ; Draw first purge line
;G1 X2.4 Y160 Z0.2 F1500.0 ; Move to side
;G1 X2.4 Y45 Z0.2 F1500.0 E40 ; Draw second purge line
;G1 X2 Y45 Z0.2 F1500.0 ; Move to purge line start position
;G1 X2 Y160 F1500.0 E20 ; Draw first purge line
;G1 X2.4 Y160 F1500.0 ; Move to side
;G1 X2.4 Y45 F1500.0 E40 ; Draw second purge line
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
;
G92 E0
G92 E0
**G1 F4200 E-5**
;LAYER_COUNT:72
;LAYER:0
M107
G1 F300 Z0.5
G0 F1800 X104.001 Y98.335 Z0.5
;TYPE:SKIRT
G1 F300 Z0.3
G1 F4200 E0.08315

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[1 Korpus.TXT](https://github.com/Ultimaker/Cura/files/7960703/1.Korpus.TXT)

closed 2022-01-28 by Berndzlin

Ghostkeeper, 2022-02-14:
Looks like a normal retraction for the initial move there. Shouldn't be a problem, right?

Berndzlin, 2022-02-14:
Der Rückzug wäre kein Problem wenn vor dem Druck  wieder ein Vorschub mit dem gleichen Wert geben würde --Diese Nachricht wurde von meinem Android Mobiltelefon mit 1&1 Mail gesendet.

Ghostkeeper, 2022-02-26:
Sorry, we communicate in English here. I'm not sure if I understand your issue properly at the moment. Could you translate it, so that the world can read along instead of just Germany and its surroundings?

It should be using the Retraction Retract Speed setting for that retraction. Your retraction speed is set to 70mm/s, which is 4200mm/min, so the speed there seems to be correct.

Berndzlin, 2022-02-27:
As can be seen from the Gcode, 5mm are drawn in at the start (line 15).
Then the print head moves to the start position for layer 0 (line 20).
Wouldn't the filament have to be advanced again by 5 mm at the latest now (line 21).
Instead, the printer starts to print the first SKIRT (line 23) with the filament feed E0.08315.

So here the step for the reverse collection is missing.

1	;FLAVOR:Marlin
2	;TIME:8257
3	;Filament used: 3.81035m
4	;Layer height: 0.2
5	;MINX:95.5
6	;MINY:88.377
7	;MINZ:0.3
8	;MAXX:159.499
9	;MAXY:152.081
10	;MAXZ:14.5
11	;Generated with Cura_SteamEngine 4.13.0
12 	M82 ;absolute extrusion mode
13	G92 E0
14	G92 E0
15	G1 F4200 E-5
16	;LAYER_COUNT:72
17	;LAYER:0
18	M107
19	G1 F300 Z0.5
20	G0 F1800 X104.001 Y98.335 Z0.5
21	;TYPE:SKIRT
22	G1 F300 Z0.3
23	G1 F4200 E0.08315

64bittuning, 2022-03-05:
@Ghostkeeper 
It seems as though the issue they are having, is that Cura is commanding a retraction prior to layer 0 beginning. Normally something like this is somewhat hidden in a skirt or brim, especially as a few mm will only create a very small gap at the start of the first layer. However this will cause significant quality issues if someone does not apply some manner of bed adhesion.

Ghostkeeper, 2022-03-14:
Cura assumes that the filament is at the nozzle tip (E0) when the start g-code is finished. As such, it must retract the filament to prevent oozing all over the build plate, which would leave a scar over the build plate. Is it a problem to retract there?

64bittuning, 2022-03-14:
Itbses to me that the code is retracting, but not unretracting before starting layer 0

Ghostkeeper, 2022-04-02:
Ah, yes I can see that happening indeed. However when I try this myself I'm not seeing this issue. These are the reproduction steps I've tried.

1. Add a Tronxy XY-2 Pro/Pro 2E/XY3 SE printer.
2. Add a cube and resize it to 100x100x10mm (to simulate it being an issue by unretracting during a 100mm line).
3. Set the retraction distance to 5mm.
4. Slice, save the g-code and inspect it.

This is the result when it starts printing the skirt:

```
G92 E0
G92 E0
G1 F2700 E-5
;LAYER_COUNT:50
;LAYER:0
M107
G1 F300 Z0.4
G0 F2700 X69.698 Y69.764 Z0.4
;TYPE:SKIRT
G1 F300 Z0.2
G1 F2700 E0
G1 F1994.5 X70.386 Y69.127 E0.02814
G1 X71.126 Y68.551 E0.05629
```

So it seems to properly put the `G1 F2700 E0` in there, to unretract.

Maybe we'd need a project file to be able to reproduce this. See if anything else is preventing that unretraction?