Ultimaker/Cura #11367: Submitting a PR to update profiles always ends in a failure

64bittuning, 2022-01-28:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Tronxy

### Reproduction steps

I have been working on an update for the Tronxy profiles currently in Cura. Upon opening the PR, submitting all of the changed files, and waiting for the build to finish, the PR errors out with a "failure" in test  mode 14/15. I have stripped my PR down to uploading a single file which has had only a few lines removed (part of the original update) yet this simple and small change continues to throw an error with a python script checker? 

I dont see how all of the plugins and checking those, has anything to do with a printer profile update, but thats why I'm submitting the bug.

I believe this may be related to #11196 
Can this be confirmed so I can upload the rest of the file package without driving myself crazy?!?

### Actual results

14/15 Test  #1: code-style .......................***Failed   21.45 sec

### Expected results

14/15 Test  #1: code-style .......................   Passed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

see #11366  and (closed) #1136

64bittuning, 2022-02-01:
Unfortunately I haven't got a single clue how to initiate the test string locally, otherwise I absolutely would do that. If there is a page I can refer to or a list of instructions you can give, that would be immensely helpful.

nallath, 2022-02-01:
We fixed the issues. So if you base it on the current master, it should work again.

as for the tests, you can run the tests with:
```cmake . 
make test```

closed 2022-02-01 by nallath