Ultimaker/Cura #11369: Hole horizontal expansion causes hole deformation

cicinovec, 2022-01-29:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Slice simple model with round hole with Hole horizontal expansion 0 mm or positive => result is round hole
2. Slice the same model with Hole horizontal expansion -0.1 mm (negative) => result is star shaped hole with deformed shape

### Actual results

1. Nice round hole
2. Deformed hole

### Expected results

Round hole with different diameter only

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[Helping_hand_clamp_PLA_T1.zip](https://github.com/Ultimaker/Cura/files/7963124/Helping_hand_clamp_PLA_T1.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7963125/cura.log)

GregValiant, 2022-01-29:
When I opened the STL and sliced it the results were about the same as you show in your images.
I opened the 3mf file and then exported the model as an STL so I could check it for errors.  None were found.
I cleared the build plate and brought in the STL (Cura 4.13.1).  There were no problems with the HHE at -0.05.  With the HHE at -0.2 there was a very slight inconsistency of the Inner-Walls around the hole.  At -0.4 it went away.
This is using your settings and at HHE = -0.05.
[image]

This is with HHE at -0.2
[image]

This image is at HHE -0.4.
[image]

cicinovec, 2022-01-29:
Hi @GregValiant , thank for your response. I see now the dependency

closed 2022-01-29 by cicinovec