Ultimaker/Cura #11370: Spiralize Comb Failure

mattwn, 2022-01-29:
### Application Version

4.13.0

### Platform

Windows 11

### Printer

Ender 3

### Reproduction steps

[image]

- Select a default profile
- Add an STL that is spiralized for most of it's height but requires a few mm of support at the bottom
- Enable Spiralize and Supports
- Slice and Preview

### Actual results

For the few layers that have the supports, combing does not work. The print head travels through the vase.

### Expected results

For the few layers that have support, the print head should comb properly and avoid the interior of the vase.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project File:
[CE3_VikingHorn.zip](https://github.com/Ultimaker/Cura/files/7963807/CE3_VikingHorn.zip)

- Other methods to produce a vase like part work well, but suffer from Z seam. Spiralize is preferred for 95% of the model. 
- Model can not be printed upside down because closed tip is unsupported and prints poorly.

smartavionics, 2022-01-30:
Like this... ?

[image]

[image]

I won't be submitting a PR and these diffs are based on my Cura build so they probably won't apply directly to the UM version but I don't mind if someone else mods the UM version along similar lines (and claims the $500, of course :))

```
diff --git a/src/FffGcodeWriter.cpp b/src/FffGcodeWriter.cpp
index fbc7afabe..3009a4e96 100644
--- a/src/FffGcodeWriter.cpp
+++ b/src/FffGcodeWriter.cpp
@@ -3227,6 +3227,22 @@ bool FffGcodeWriter::processSupportInfill(const SliceDataStorage& storage, Layer
 
             if (all_insets.size() > 0)
             {
+                if (infill_extruder.settings.get<bool>("magic_spiralize"))
+                {
+                    size_t layer_nr = gcode_layer.getLayerNr();
+                    if(layer_nr > infill_extruder.settings.get<size_t>("bottom_layers"))
+                    {
+                        // bit of subtlety here... support is being used on a spiralized model and to ensure the travel move from the end of the last spiral
+                        // to the start of the support does not go through the model we have to tell the slicer what the current location of the nozzle is
+                        // by adding a travel move to the end vertex of the last spiral. Of course, if the slicer could track the final location on the previous
+                        // layer then this wouldn't be necessary but that's not done due to the multi-threading.
+                        const Polygons* last_wall_outline = storage.spiralize_wall_outlines[layer_nr - 1];
+                        if (last_wall_outline != nullptr)
+                        {
+                            gcode_layer.addTravel((*last_wall_outline)[0][storage.spiralize_seam_vertex_indices[layer_nr - 1]]);
+                        }
+                    }
+                }
                 setExtruder_addPrime(storage, gcode_layer, extruder_nr); // only switch extruder if we're sure we're going to switch
                 gcode_layer.setIsInside(false); // going to print stuff outside print object, i.e. support
                 gcode_layer.addPolygonsByOptimizer(all_insets, gcode_layer.configs_storage.support_infill_config[0]);
diff --git a/src/LayerPlan.cpp b/src/LayerPlan.cpp
index 0b3ec47d7..eacc2cdb5 100644
--- a/src/LayerPlan.cpp
+++ b/src/LayerPlan.cpp
@@ -1866,7 +1866,9 @@ void LayerPlan::spiralizeWallSlice(const GCodePathConfig& config, ConstPolygonRe
     // once we are into the spiral we always start at the end point of the last layer (if any)
     // NOTE: this travel move is now ignored when the g-code is being generated but we can't actually remove it
     const Point origin = (last_seam_vertex_idx >= 0 && !is_bottom_layer) ? last_wall[last_seam_vertex_idx] : wall[seam_vertex_idx];
-    addTravel_simple(origin);
+    // FURTHER NOTE: this used to use addTravel_simple() but if support is being used then combed travel should be used to avoid the nozzle crossing the model
+    // on its return from printing the support.
+    addTravel(origin);
 
     if (!smooth_contours && last_seam_vertex_idx >= 0) {
         // when not smoothing, we get to the (unchanged) outline for this layer as quickly as possible so that the remainder of the
```

smartavionics, 2022-01-30:
I forgot to say that I had to change the combing mode to within infill for it to avoid all travel within the model.

GregValiant, 2022-01-30:
But the bottom will fail as the layer lines are too far apart to connect?  There are some large gaps there.  The image below is a slice with no changes to the 3mf file.  Even at 0.1 layer height the first few layers would be unattached from each other.
[image]

Here's my best shot with 4.13.1

[image]
[image]

Here's the 3mf.  Let me know if I win.  (I don't really see a bug here.)
[GV_VikingHorn.zip](https://github.com/Ultimaker/Cura/files/7966273/GV_VikingHorn.zip)

mattwn, 2022-01-30:
Hi Greg, yes, the model might not be connected at the tip. This was a super simple demo of a range of vases we make. We typically slice at 0.8 z height. We also use fuzzy skin frequently. I left it simple so it would slice quickly, because this specific model is not really the issue, it's the combing behavior between `;TYPE:SUPPORT` and `;TYPE:OUTER-WALL` when spiralize is enabled.

You 3mf still has the same issue. It looked fine because you had the z seam aligned with the support, but if I rotate the z seam, the offending travels are still present, no combing happened. Some of our models have supports on both sides, so aligning the z-seam like this is not always an solution for us.

Here is a screenshot of your 3mf with the z seam rotated.
[image]

mattwn, 2022-01-30:
Hi SmartAvionics. That sure looks like promising! I see the z seam is unaligned with the supports and there are still no offending travel lines through the interior. Fantastic.

Since you only modified FffGcodeWriter.cpp, I assume this is just an impact to CuraEngine? If you could send us a built Windows compatible CureEngine.exe with this fix that works with mainline Cura Front End, we'd be happy to send you half of the bounty and then send the other half to whoever can integrate it into mainline CuraEngine.

smartavionics, 2022-01-30:
I'll try and build an upstream 4.13.1 CuraEngine.exe with these changes but if I have a problem doing that, I'll just build one of my Windows releases and you will have to try using that.

mattwn, 2022-01-30:
If it's easier to just build your custom Front End and Engine (with the fix) together, we can use that in the mean time and send you half the bounty (assuming your customizations don't result in other unforseen problems =)! ) Whatever is easiest for you. Not trying to add to your workload. Just happy to pay out and show our gratitude if it works. Assuming it works, send me a private message with how you would like to be paid. Zelle is preferable on our end.

We'll then still reserve half the bounty for whoever can eventually get the fix rolled into mainline.

smartavionics, 2022-01-30:
I've build an engine for you, you can download it from

https://github.com/smartavionics/Cura/releases/download/20220129/CuraEngine-for-mattwn.exe

Just rename to CuraEngine.exe and drop it into the 4.13.1 installation.

If you have any feedback, please open an issue at https://github.com/smartavionics/Cura/issues rather than commenting here.

And, I'm really not interested in payment but I do appreciate the offer.

mattwn, 2022-01-30:
Thanks! We emailed a Zelle payment link. I know you say you're not interested, but it's there if you want it.

It worked great. We did notice a slicing failure when adding a raft. But I reported it in your Repo as requested. We'll continue any conversation there.

The other half of the bug bounty is still there if anyone is able to eventually roll this fix into CuraEngine mainline.