Ultimaker/Cura #11371: Surface Mode is oversized by 1 wall thickness

mattwn, 2022-01-29:
### Application Version

4.13.0

### Platform

Windows 11

### Printer

Ender 3

### Reproduction steps

[image]

1. Select a default profile
2. Load any STL
3. place a cutting mesh at the bottom, only change is `Surface Mode: Surface`
4. Slice

### Actual results

A step change equal to 1 wall thickness is observed at the transition from the `Surface Mode:Surface` cutting mesh to the rest of the model.

### Expected results

`Surface Mode: Surface` region would be the exact size as the normal slice. 

In the specific example above, no step change in size should be observed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project File:
[CE3_VikingHorn-SurfaceSizeBug.zip](https://github.com/Ultimaker/Cura/files/7963869/CE3_VikingHorn-SurfaceSizeBug.zip)

- Attempted many permutations of `Slicing Tolerance` between the cutting mesh and the global, but step change was always present. I believe Slicing Tolerance is not respected per mesh in any case.
- Attempted to apply negative `Horizontal Expansion` of 1 wall thickness to the cutting mesh, but step change was still present. I believe `Horizontal Expansion` is not respected per mesh.
- Concern is not necessarily that this produces a step change with a cutting mesh, that is just a method to demonstrate that `Surface Mode: Surface` produces slices oversized by 1 wall thickness

fvrmr, 2022-02-08:
I have discussed it with the team and decided to defer this issue, since it is a very edge case to use surface mode with a cutting mesh. I hope you understand.

Ghostkeeper, 2022-02-14:
With surface mode, Cura simply prints the surface. This has a number of advantages, not least among which is that it can print broken models where it's impossible to determine where the inside is. It simply prints every triangle, even if the mesh is not water tight.

Normally, Cura prints the walls such that they are contained in the shape. In surface mode, it prints the walls with the centreline exactly on the surface, causing the surface to protrude by half the line width in both directions. That is what you're seeing here. It's a half-line-width protrusion.

Slicing Tolerance and Horizontal Expansion have no effect because Cura makes no assumptions any more on what the inside is of the model and what the outside is.

I suggest, if you simply want to print the surface with a single wall, you set the Wall Line Count to 1. That is what the setting is for, in the end. To get a similar result to Surface Mode, you would also have to set the Top/Bottom Thickness to 0mm.

closed 2022-02-14 by Ghostkeeper