Ultimaker/Cura #11373: Gap in Preview mode

chiles13, 2022-01-29:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Anet 8

### Reproduction steps

1. Imported an .stl file for filament filer. Looks perfect on the bed.
2. Sliced the file with/without support, Still looks fine in prepare mode.
3. Looked at it in Preview mode and part of the print is missing.

### Actual results

When running the print in Preview mode, the area, which is a bridge between two sides, does not print at all.
Also the support for the same area does not show up, but it does show in other areas.

### Expected results

In Preview mode I should be able to see both the support and the bridge.
Obviously when printed this bridge and support do not print and so the filter does not clip together.
The last time the .stl file was used in 2016, and it printed fine then. I am still using one of the two printed filters.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Filla Filter(1).zip](https://github.com/Ultimaker/Cura/files/7964028/Filla.Filter.1.zip)
[image]
[image]

Asterchades, 2022-01-29:
Your model is faulty. There's a visible outline of the piece you used to fill in the bridge, and only the outer edge of said bridge is marked as being overhang. This is also likely responsible for the unevenness of the top of the near wall, which appears to be a layer shorter going around the corners.

Correct the model - either manually or with an automated tool (eg MeshMixer or 3D Builder) - and it should preview and print as you expect. You might also be able to get it work with a combination of Mesh Fixes within Cura, or the Mesh Tools plugin, but that will require more experimentation on your part.

chiles13, 2022-01-30:
HiMany thanks for your reply. I done a repair in 3D builder and also removed the bridge I put in to try and fill the gap. I also upgraded to V4.13.1 and tried again but was unsuccessful.However, after upgrading I decided to add a new basic profile and changed a few of my preferred settings. It worked and now displays the bridge and support.I now just need to go through what setting I have added to my favoured profile that is causing the missing bridge and support.RegardsMick

GregValiant, 2022-01-30:
It may have been the Horizontal Expansion of "-0.19".  The wall is 0.8mm thick but that bridge was only 0.7mm thick.  Taking .19 off both sides of the bridge made it 0.32 thick and your line width of 0.4 was too much and so the bridge wouldn't slice.  At a Horizontal Expansion of -0.14 it did slice.

chiles13, 2022-01-30:
Hi GregThanks for the reply.That's very interestng, I will focus on that.RegardsMick

fvrmr, 2022-01-31:
Hi @chiles13 thank you for your report.
Like @Asterchades mentioned your model is not watertight. And also @GregValiant is right that if you change the Horizontal expansion to 0.14 or 0 it will slice correctly.

I will close this issue, but reach out if you have more questions.

closed 2022-01-31 by fvrmr

chiles13, 2022-01-31:
Hi 
Thank you for your help.As suggested, I set the horizontal expansion back to 0 and it sliced with support.I am very happy.RegatrdsMick