Ultimaker/Cura #11374: Wall support filter

Asterchades, 2022-01-29:
### Is your feature request related to a problem?

While not specifically a problem, in many situations the use of blanket supports isn't strictly necessary. By using the already known properties of a given slice in tandem with the capability of bridging between level points it should be possible to reduce the time and material spent on supports, while also potentially making them easier to remove.

### Describe the solution you'd like

My suggestion is to optionally adapt the support structure to only take into consideration where the walls of layers requiring support would be, rather than the skin as well. Naturally walls themselves don't tend to be very thick so this would need to have a larger coverage to permit stable support structures, either of a fixed or user-configurable distance.

If you were to take the basic situation of an inverted L and apply this, instead of having a solid block of supports (density withstanding) under the overhanging "arm" it would instead have more of a square bracket shape ("[") of supports with a hollow in the centre.

### Describe alternatives you've considered

An alternative would be similar to the (currently experimental) feature within PrusaSlicer which does not apply supports to bridges. This has the potential to use even less material and time but I feel would be more computationally expensive as this requires determining what is or isn't a bridge, rather than using the already-known location of walls. _Though this could potentially be a long-term expansion of the feature._

The current support implementation also has facilities to use a number of support walls, but these only support the outermost portion of any given area of the model, potentially leaving parts of the model within this area completely unsupported. It would also result in a fixed 100% density for any supports which are generated as they would just be walls, rather than the more typical infill-style supports.

Concentric suffers from similar limitations in that it takes no consideration of the geometry within the outer bounds - only the shape of the boundary itself. Without any form of adaptive density spacing it will invariably miss parts, and again its lines can only be done effectively solid.

### Affected users and/or printers

Anyone attempting to print an object which doesn't necessarily require a full suite of supports but still has some areas that need propping up. It naturally will not print with the same overall quality as having the supports there due to the extra spacing for bridges, but not everything needs to be perfectly clean on the bottom - sometimes it just needs to be present.

fvrmr, 2022-02-04:
Hi @Asterchades thank you for your feature request.
I will bring this up for discussion with the team. Keep you posted!

MariskaMaas, 2022-02-04:
Hey Asterchades!

I'm trying to understand what you are looking for and how it is different from what currently is offered in Cura.

Are you trying to achieve Option A?
You get this when you tune the Support Wall Line Count and Support Density in concentric support.
[image]

Or Option B?
You get this when you enable Tree support. It already offers single-walled support.
[image]

If you mean something different altogether, can you please supply a drawing or visualization of some kind? 
Thank you!

Asterchades, 2022-02-04:
Similar to option A, but not quite.

Forgive the pictures. Figured SketchUp was the best tool I have at my disposal, but it's still not that good. Consider that we're using the same basic shape - an inverted L, 20mm x 20mm on the beams and 60mm tall. Theoretical support settings involve the use of 3mm width and no horizontal expansion.

[image]
This is what I would expect just applying a support to where the walls are. It looks almost identical to Option A, save for the absence of the wall up against the main structure on the left. Splitting hairs in this specific example, perhaps, but there's more.

[image]
This is the same inverted L, only now with an 8mm x 8mm x 5mm (though the depth is irrelevant, so long as it doesn't reach the bed) protrusion on the bottom. Because this protrusion has its own walls, support can be present underneath it as well - complete with a 2mm x 2mm hole in the middle due to the specified width.

[image]
This is the same inverted L with protrusion as sliced by Cura 4.13.0 (4.13.1 should be identical, as should Arachne) to replicate Option A, using 0% density and eight walls (3.2mm), as viewed from the layer directly under the protrusion. In addition to the wall against the main structure on the left, there's now superfluous supports running up the sides of the protrusion - but there's absolutely nothing directly underneath it. Further these walls are completely solid.

[image]
Alternatively, if the protrusion were instead a 5mm recess, my proposed method would filter the supports based on the presence of the walls around the recess. This leaves the corners of the opening at the mouth of the recess well supported, theoretically allowing both the indent and the overhang to print completely.

[image]
Where-as with Option A, this once again creates a superfluous structure that starts in mid air which would only affect the top of the recess - assuming it even managed to stick to something by accident so it could hang there. The corners of the overhang itself would fail, though much of the overhang would still succeed in this instance as it is only a short bridge. NB If the recess becomes a hole all the way through then the two methods once again come closer to each other.

I don't mean for the support structures to be solid in my idea. They would use the user-specified pattern. I'm just not about to go trying to model that in SketchUp, 'cause I don't have enough sanity left to survive that ordeal.

fvrmr, 2022-02-08:
Thank you for your reply, we have discussed this with the team and decided to defer this request, it will not get a high priority for us to implement. I hope you understand.

Asterchades, 2022-02-08:
All good, chief. Just something to have a think about in the background.

Ghostkeeper, 2022-02-14:
Sounds a bit like Skin Edge Support, but for support instead of infill. Actually makes a lot of sense putting it that way.

Asterchades, 2022-02-14:
If you'll permit me a Genie moment: Boy do I feel sheepish. That would have been a much better description.

Though instead of having a user-defined height it would instead have a user-defined width, since it's always going to be tall enough to reach the bed/model beneath it and single walls aren't exactly stable when in long, straight lines.

closed 2023-07-13 by github-actions[bot]