Ultimaker/Cura #11376: 윈도우11 에서 실행시  프리징현상을 보입니다

bushse, 2022-01-29:
### Application Version

4.13.1

### Platform

windows11

### Printer

clone s-290

### Reproduction steps

정상적인큐라 설치 후 셋팅

### Actual results

큐라 실행시 프리징현상(응답없음)이 보임
특히 구성의 재료설정 에서 증상이 많이 나타남
버전을 바꾸어 설치 해봐도 동일함 
4.10 이전 버전설치시 응답없음 시간이 약간 줄어듬

### Expected results

[cura.log](https://github.com/Ultimaker/Cura/files/7964868/cura.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2022-01-31:
Hi @bushse thank you for your report.
Please fill in the form in English.

Ghostkeeper, 2022-03-14:
Looks like a duplicate of https://github.com/Ultimaker/Cura/issues/11381 then. Please continue the conversation there!

closed 2022-03-14 by Ghostkeeper