Ultimaker/Cura #11377: Cannot set Build Plate Adhesion Type

RCPDesigns, 2022-01-29:
### Application Version

4.13.1

### Platform

MacOS

### Printer

LK5 Pro

### Reproduction steps

Open Preferences, select Settings, search for and/or scroll to Build Plate Adhesion Type.  

### Actual results

Note that the Build Plate Adhesion Type is a checkbox, type a drop down to select from Skirt, Brim, Raft.

### Expected results

Build Plate Adhesion Type should be a drop down to select from Skirt, Brim or Raft.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-01-30:
Is that the "Settings Visibility" dialog?  When that box is checked then the drop down for Build Plate Adhesion Type should appear in the main settings list on the right side of the workspace under the heading "Build Plate Adhesion".

RCPDesigns, 2022-01-30:
Yes, that is correct so I guess this is user error on my part.  I will say,
however, as a Mac user I expect Settings/Preferences to be in a
specific place, not the control of the visibility of the settings somewhere
else.  In addition, even with your explanation it took me some time to
notice a very faint down arrow at the far right side.  That expanded to yet
another settings list that, again, had a check mark for Build Plate
Adhesion.   Clicking on "Custom" brings up more settings where the Build
Plate Adhesion can be selected.   I completely understand that given the
number of settings there needed to be some way to reduce what is visible.
Having visibility settings is a great idea.  Putting it where, for a Mac
user at least, settings/prefs lives is not the best idea.

GregValiant, 2022-01-31:
This is one of the downsides of building a piece of software for multiple operating systems.  I'm on Windows, you're on a Mac, and I think many of the developers use Linux.  They are at the mercy of the software used to make the program cross-platform.
Back when I worked in tool design shops we had a couple of sayings:
"Ambidextrous means it won't work exactly right for anything."
and the related
"Universal means you'll have to get out a cutting torch and a sledge hammer."

Above all the settings is the "Settings Search" box.  To the right of that is an icon with three lines on it.  It is the "Settings Visibility" tool.  Click on it and set the visibility to "All".  It's one of the first things I do when I start a session so I don't miss anything.

fvrmr, 2022-01-31:
Hi @RCPDesigns thank you for your report.
In the preferences you select what materials you would like to have visible in the printer settings. In the printer settings on your right you can change the setting to your desired value, for example the build plate adhesion type to brim. 
This is the same for every operating system.

RCPDesigns, 2022-01-31:
Fenne,

Thank you for your response.  Greg cleared this up over the weekend.  The
software is operating correctly just not how I expected it to be.

RCPDesigns, 2022-01-31:
As a programmer that delivers to multiple platforms I can certainly relate.
  We chose to do our UI native to each platform.  At times we wish we had a
unified UI.

Cura is outstanding software and has been easy to work with.

fvrmr, 2022-02-01:
Glad to hear! I will close the issue then since it is working as expected!

closed 2022-02-01 by fvrmr