Ultimaker/Cura #11380: Tree Support Extra Prime Amount Setting

Deadly-Cicada, 2022-01-30:
### Is your feature request related to a problem?

Printer/settings:
Ender 3 w/ Titan Aero direct drive
hatchbox 1.75mm white PETG
250 C hotend
75 C bed temp

print speed: 60mm/s
wall speed inner/outer: 30mm/s
travel speed: 150mm/s

When printing tree supports; new layers often do not have enough filament primed in the nozzle and create gaps in support.

Seems to be related to [GR5's answer](https://community.ultimaker.com/topic/29121-underextrusion-on-outer-wall/?do=findComment&comment=289506) to a question regarding general under-extrusion where the printer is moving from an inner wall that is very thin and there's very little filament in the nozzle > retracting > traveling > "starting" > under-extruding

Image shows large mask being printed and all tree supports on the left side of the print (where layers started) were under extruded and weak.

[image]
g

attempted another print after this one with retraction turned off and "minimum extrusion distance window" changed from 10mm to 1mm to match the retraction distance. No change in support quality. Problem is happening on smaller models with tree supports as well. symptoms less pronounced but still present.

### Describe the solution you'd like

Suggest: a setting that allows one to prime extra filament into the nozzle; specifically for tree supports so that they can ensure the correct amount even when traveling from thin inner walls to the beginning of the tree support. 

Could be placed in Support or Travel setting under "Tree Support Extra Prime Amount"  - _Nozzle may not be primed with enough material to begin a tree support, which can be compensated for here._

### Describe alternatives you've considered

alternatively there could be a setting for an extra prime amount when transitioning from any wall thinner than X-amount to any other wall, support structure, etc. which allows for priming in relation to the previously extruded amount and not necessarily related to retraction distance alone.

This could also be a hidden setting calculated automatically using the difference between the end of the previous layer to the necessary amount of the next layer or support. 

### Affected users and/or printers

This would benefit any user experiencing under-extrusion on tree supports and possibly (if alternative is implemented) reduce problems with under-extrusion between layer end and layer start where the printer is transitioning from a very small thin layer to a larger standard layer.

Asterchades, 2022-01-30:
Does un-checking Limit Support Retractions have any influence on the result?

Deadly-Cicada, 2022-01-30:
> Does un-checking Limit Support Retractions have any influence on the result?

Need to test. My thought would be that this would have no effect, or a negative effect. Whenever the under-extrusion was happening, the nozzle was traveling directly from the object to the support not from the support to another support. that, and this would enable retractions between supports which would create an even larger gap in the next layer to be printed.

Ghostkeeper, 2022-02-14:
This seems to be an example of the more general problem of having a flow change from low to high flow. Travel moves (retracted or no) in between have surprisingly little impact on flow changes, in my experience. I think this is not specific to tree support (why would it?) but just for flow changes.

We have a proposed change that would compensate for this: https://github.com/Ultimaker/CuraEngine/tree/flow_advance . It adjusts for flow changes in advance. It needs a bit of love to make sure the result doesn't look too weird in layer view though.

closed 2022-02-17 by no-response[bot]