Ultimaker/Cura #11381: Crashes when accessing materials and profiles

NebTrebmal, 2022-01-30:
### Application Version

4.13

### Platform

Windows 10

### Printer

Ender 5 Pro

### Reproduction steps

1. Had a model loaded 
2. Changed material to eSun PLA + White

### Actual results

1. Profile changed from my 'PLA Standard profile' to 'Not supported'
2. Now I can't access any of my material and profile settings - Cura freezes every time I try. 
3. Even if I change material from eSun PLA, Cura still crashes

### Expected results

I should have been able to continue onto the slicing process and then onto printing as normal

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7965840/cura.log)

fvrmr, 2022-02-04:
Hi @NebTrebmal thank you for your report.
Your log file doesn't show us why or when it is crashing. It could be a duplicate of https://github.com/Ultimaker/Cura/issues/9861
There a workaround was to run Cura as an administrator.

GitOutaTown, 2022-02-19:
After many crashes and re-installs I have so far traced this as far as the XML.FDM_Material Metadata.

Example  --->   eSUN_PLA_PRO_Black.xml.fdm_material

Changing 

    <metadata>
        <name>
            <brand>eSUN</brand>
            <material>PLA+</material>                 <<<<<< this appears to be the offending parameter
            ...

to 

    <metadata>
        <name>
            <brand>eSUN</brand>
            <material>PLA</material>                      <<<<<<  Remove the "+" Character stops the crashing
            ...

appears to fix the problem.

I have tried other phrases like PLAPlus , PLAPro ETC. and they all crash the program ...  It seems that having a "material" value that isn't within a specified group of values is the issue.

fvrmr, 2022-02-22:
This should not crash Cura. But what could be happening is that when having a material name like PLAPlus it will not show supported profiles, because PLAPlus is an unknown material type. So indeed when changing the name to PLA is will show the profiles that are compatible with material. See https://github.com/Ultimaker/Cura/issues/8229

Ghostkeeper, 2022-03-07:
> Now I can't access any of my material and profile settings - Cura freezes every time I try.

How are you trying this? Is it the opening of the drop-down menu that causes this? Or are you doing something else than simply opening the drop-down?

64bittuning, 2022-03-10:
@fvrmr 
@Ghostkeeper 

This issue is not only limited to this one machine. I am fighting with something similar at the moment and seem to have tracked the issue down to all materials with a non-standard <material>"name"</material> listed in their  fdm_material XML file. 

Another thing worth noting is that this fault is also present in CPE+, Tough PLA, and a few other instances.

Please view
https://github.com/Ultimaker/fdm_materials/pull/196

bushse, 2022-03-25:
> 안녕하세요 @NebTrebmal 귀하의 보고서에 감사드립니다. 로그 파일은 왜 또는 충돌하는 시기를 알려주지 않습니다. 그것은 #9861 의 중복 될 수 있습니다 해결 방법이 관리자로 큐라를 실행하는 것이었습니다.

bushse, 2022-03-25:
nesteuwokeu yeongyeol-eul kkeunhgo hamyeon sesting-ui jaelyoseoljeong bubun-ui meomchum hyeonsang-eun haegyeol doena seullaising hu milibogi dangyeui jeongjihyeonsang-eun haegyeol-i anhdoenda